zephyrproject-rtos/zephyr #31259: uart.h: Clarification required on uart_irq_tx_ready uart_irq_rx_ready

erwango, 2021-01-12:
This point relates to discussion that happened around https://github.com/zephyrproject-rtos/zephyr/pull/31192.

It appears that there are some uart API clients (in tree: h4.c and more, @jfischer-no may complete the list) that rely on a specific behavior from `uart_irq_tx_ready` and `uart_irq_rx_ready`.

The behavior variant, not explicited in uart.h, is that functions return false if matching IRQ are disabled.
It is implemented in MCUX and NRF uart drivers (and now STM32, following merge of #31192, maybe few others), but not in other uart drivers (nuovoton, sam, sam0, ..).

So question is:
- Should this behavior be explicit in API and updated in drivers that don't behave this way ?
- Should it be reverted and clients fixed ?
- Should it be kept optional but made explicit in the API ?

erwango, 2021-01-12:
^^ @carlescufi  @jfischer-no @dcpleung

jfischer-no, 2021-01-12:
The API does not provide separate callbacks for TX and RX events. Construct like in the example below can be found in several places. Assuming that after `uart_irq_tx_disable(ctx->dev);`,  `uart_irq_tx_ready(ctx->dev)` returns false, one can avoid the use of additional controls.

```c
static void uart_callback(const struct device *dev, void *user_data)
{
    ....
	while (uart_irq_update(ctx->dev) && uart_irq_is_pending(ctx->dev)) {
                if (uart_irq_rx_ready(ctx->dev)) {
			foo_handler_rx(ctx);
		}

                if (uart_irq_tx_ready(ctx->dev)) {
			foo_handler_tx(ctx);
		}
	}
    ....
}
```

- [drivers/bluetooth/hci/h4.c](https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/bluetooth/hci/h4.c#L442-L450)
- [subsys/mgmt/osdp/src/osdp.c](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/mgmt/osdp/src/osdp.c#L74-L91)
- [subsys/net/lib/openthread/platform/uart.c](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/net/lib/openthread/platform/uart.c#L108-L117)
- https://github.com/zephyrproject-rtos/zephyr/pull/26516/files#diff-a994711763bf9047172526af8656a1817ed7b2191b8e3d5e358527bedff18165R445-R454

carlescufi, 2021-01-12:
API meeting 12th January 2021:

No decision taken, but alternatives seem to be:

a) Document that the functions perform a logical AND between the rx/tx flags and the rx/tx IRQ enable flags
b) Do not perform the logical AND above and instead provide the functions below:
```
bool uart_irq_rx_is_enabled()
bool uart_irq_tx_is_enabled()
```

a) would probably require no changes to API users in-tree, whereas b) would require some users to perform:

```
if (uart_irq_tx_ready(h4_dev) && uart_irq_rx_is_enabled(h4_dev)) {
```

Would very much appreciate @nordic-krch and @Mierunski input on this particular decision.

pabigot, 2021-01-12:
Note that the [existing documentation](https://docs.zephyrproject.org/latest/reference/peripherals/uart.html#group__uart__interface_1gad04073b1b8e3de13b43ae1194561377b) very clearly states that `uart_irq_rx_ready` indicates that there is unreceived data pending, which is independent of whether IRQ is still enabled when the function is invoked.  There are use cases where something may disable the interrupt before the handler can be invoked, in which case the additional check would cause `uart_irq_rx_ready()` to fail to meet its specification (that there's data ready).  Also in that case a separate function to detect IRQ enable state would not help as there's a race condition with the receive buffer check.

I think we need to go back to the time when the additional gate was first added in Nordic and understand why that was done, and figure out whether there's a better way to address it that doesn't change the documented behavior.

I have a list of issues with the UART API that this relates to, along with some draft samples showing how to use the three different UART APIs.  I will try to dust those off which might help clarify questions of how to use the API correctly.

nordic-krch, 2021-01-14:
As for nordic drivers, `uart_irq_rx_ready` returns true if there is any pending data, no check for interrupt being enabled.

As for `uart_irq_tx_ready`, it checks against Tx interrupt being enabled but API states:
```
@brief Check if UART TX buffer can accept a new char
```
As it can only be called from uart interrupt context, then it seems ok to not accept new data when interrupt is disabled. Given that `uart_irq_tx_disable` is also intended to be called from uart interrupt context.

carlescufi, 2021-01-14:
> It is implemented in MCUX and NRF uart drivers (and now STM32, following merge of #31192, maybe few others), but not in other uart drivers (nuovoton, sam, sam0, ..).

@erwango this seems to be inaccurate as per [this comment](https://github.com/zephyrproject-rtos/zephyr/issues/31259#issuecomment-760181612) on the RX side. On the TX it is, but see the reasoning for it in the same comment.

erwango, 2021-01-14:
> @erwango this seems to be inaccurate as per this comment on the RX side.

Exact. >till stands for `uart_mcux.c` though.

carlescufi, 2021-01-14:
> > @erwango this seems to be inaccurate as per this comment on the RX side.
> 
> Exact. >till stands for `uart_mcux.c` though.

Then this is instead a bug in `uart_mcux.c` (and now in the ST UART driver?)

erwango, 2021-01-15:
> > > @erwango this seems to be inaccurate as per this comment on the RX side.
> > 
> > 
> > Exact. >till stands for `uart_mcux.c` though.
> 
> Then this is instead a bug in `uart_mcux.c` (and now in the ST UART driver?)

ok, will fix STM32 one.
@MaureenHelm can you check `uart_mcux.c` ?

carlescufi, 2021-01-19:
This is now a bug in `uart_mcux.c`.

PR #31843 (2021-02-03): drivers: serial: Don't condition uart_irq_rx_ready on irq enabled
The function uart_irq_rx_ready() should return true if there is data in
the receive buffer, regardless of whether the irq is enabled. Fix the
mcux and rv32m1 shim drivers to implement this behavior correctly.

Fixes #31259
closed 2021-02-03 by nashif