zephyrproject-rtos/zephyr #31272: CANOpen Sample compilation fails

Nukersson, 2021-01-12:
**Describe the bug**
Compiling `samples/subsys/canbus/canopen` in master fa6ac271ec9f4c5430e426f2e62b4382147868cf fails with error.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\nucleo_f746zg
3. make
4. See error:
```
[71/172] Building C object zephyr/CMakeFiles/zephyr.dir/subsys/settings/src/settings_nvs.c.obj
FAILED: zephyr/CMakeFiles/zephyr.dir/subsys/settings/src/settings_nvs.c.obj 
ccache /home/<user_name>/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DBUILD_VERSION=zephyr-v2.4.0-2966-gfa6ac271ec9f -DCORE_CM7 -DHSE_VALUE=8000000 -DKERNEL -DSTM32F746xx -DUSE_FULL_LL_DRIVER -DUSE_HAL_DRIVER -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR__=1 -I../include -Izephyr/include/generated -I../soc/arm/st_stm32/stm32f7 -I../drivers -I../soc/arm/st_stm32/common/. -I../soc/arm/st_stm32/common -I../subsys/settings/include -I../subsys/canbus/canopen/. -I/home/<user_name>/Documents/DevProjects/zephyrproject/modules/hal/cmsis/CMSIS/Core/Include -I/home/<user_name>/Documents/DevProjects/zephyrproject/modules/lib/canopennode/. -I/home/<user_name>/Documents/DevProjects/zephyrproject/modules/lib/canopennode/stack -I/home/<user_name>/Documents/DevProjects/zephyrproject/modules/hal/stm32/stm32cube/stm32f7xx/soc -I/home/<user_name>/Documents/DevProjects/zephyrproject/modules/hal/stm32/stm32cube/stm32f7xx/drivers/include -I/home/<user_name>/Documents/DevProjects/zephyrproject/modules/hal/stm32/stm32cube/stm32f7xx/drivers/include/Legacy -I/home/<user_name>/Documents/DevProjects/zephyrproject/modules/hal/stm32/stm32cube/common_ll/include -I../samples/subsys/canbus/canopen/objdict -isystem ../lib/libc/minimal/include -isystem /home/<user_name>/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.2.0/include -isystem /home/<user_name>/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.2.0/include-fixed -Os -imacros /home/<user_name>/Documents/DevProjects/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mcpu=cortex-m7 -mthumb -mabi=aapcs -imacros /home/<user_name>/Documents/DevProjects/zephyrproject/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wno-address-of-packed-member -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/<user_name>/Documents/DevProjects/zephyrproject/zephyr/samples/subsys/canbus/canopen=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/<user_name>/Documents/DevProjects/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/<user_name>/Documents/DevProjects/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -nostdinc -MD -MT zephyr/CMakeFiles/zephyr.dir/subsys/settings/src/settings_nvs.c.obj -MF zephyr/CMakeFiles/zephyr.dir/subsys/settings/src/settings_nvs.c.obj.d -o zephyr/CMakeFiles/zephyr.dir/subsys/settings/src/settings_nvs.c.obj -c /home/<user_name>/Documents/DevProjects/zephyrproject/zephyr/subsys/settings/src/settings_nvs.c
In file included from ../include/arch/arm/aarch32/arch.h:20,
                 from ../include/arch/cpu.h:19,
                 from ../include/kernel_includes.h:33,
                 from ../include/kernel.h:17,
                 from ../include/fs/nvs.h:11,
                 from ../subsys/settings/include/settings/settings_nvs.h:11,
                 from /home/<user_name>/Documents/DevProjects/zephyrproject/zephyr/subsys/settings/src/settings_nvs.c:12:
/home/<user_name>/Documents/DevProjects/zephyrproject/zephyr/subsys/settings/src/settings_nvs.c: In function 'settings_backend_init':
../include/devicetree/fixed-partitions.h:53:9: error: 'DT_COMPAT_fixed_partitions_LABEL_storage_PARTITION_ID' undeclared (first use in this function)
   53 |  DT_CAT(DT_COMPAT_fixed_partitions_LABEL_, label)
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../include/devicetree.h:2127:24: note: in definition of macro 'DT_CAT'
 2127 | #define DT_CAT(a1, a2) a1 ## a2
      |                        ^~
../include/storage/flash_map.h:263:2: note: in expansion of macro 'DT_FIXED_PARTITION_ID'
  263 |  DT_FIXED_PARTITION_ID(DT_NODE_BY_FIXED_PARTITION_LABEL(label))
      |  ^~~~~~~~~~~~~~~~~~~~~
../include/devicetree/fixed-partitions.h:53:2: note: in expansion of macro 'DT_CAT'
   53 |  DT_CAT(DT_COMPAT_fixed_partitions_LABEL_, label)
      |  ^~~~~~
../include/storage/flash_map.h:263:24: note: in expansion of macro 'DT_NODE_BY_FIXED_PARTITION_LABEL'
  263 |  DT_FIXED_PARTITION_ID(DT_NODE_BY_FIXED_PARTITION_LABEL(label))
      |                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/<user_name>/Documents/DevProjects/zephyrproject/zephyr/subsys/settings/src/settings_nvs.c:272:23: note: in expansion of macro 'FLASH_AREA_ID'
  272 |  rc = flash_area_open(FLASH_AREA_ID(storage), &fa);
      |                       ^~~~~~~~~~~~~
../include/devicetree/fixed-partitions.h:53:9: note: each undeclared identifier is reported only once for each function it appears in
   53 |  DT_CAT(DT_COMPAT_fixed_partitions_LABEL_, label)
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../include/devicetree.h:2127:24: note: in definition of macro 'DT_CAT'
 2127 | #define DT_CAT(a1, a2) a1 ## a2
      |                        ^~
../include/storage/flash_map.h:263:2: note: in expansion of macro 'DT_FIXED_PARTITION_ID'
  263 |  DT_FIXED_PARTITION_ID(DT_NODE_BY_FIXED_PARTITION_LABEL(label))
      |  ^~~~~~~~~~~~~~~~~~~~~
../include/devicetree/fixed-partitions.h:53:2: note: in expansion of macro 'DT_CAT'
   53 |  DT_CAT(DT_COMPAT_fixed_partitions_LABEL_, label)
      |  ^~~~~~
../include/storage/flash_map.h:263:24: note: in expansion of macro 'DT_NODE_BY_FIXED_PARTITION_LABEL'
  263 |  DT_FIXED_PARTITION_ID(DT_NODE_BY_FIXED_PARTITION_LABEL(label))
      |                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/<user_name>/Documents/DevProjects/zephyrproject/zephyr/subsys/settings/src/settings_nvs.c:272:23: note: in expansion of macro 'FLASH_AREA_ID'
  272 |  rc = flash_area_open(FLASH_AREA_ID(storage), &fa);
      |                       ^~~~~~~~~~~~~
[84/172] Building C object zephyr/CMakeFiles/zephyr.dir/drivers/can/can_stm32.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/<user_name>/Documents/DevProjects/zephyrproject/zephyr/build
The terminal process "/usr/bin/zsh '-c', 'west build -p auto -b ${BOARD}${CPU} ${PRJ_NAME}'" terminated with exit code: 1.
```

**Expected behavior**
Compilation of the project without error.

**Impact**
Showstopper for CAN-FD implementation!

**Environment (please complete the following information):**
 - OS: Ubuntu Linux 20.04
 - Toolchain Zephyr SDK
 - Commit SHA or Version used: fa6ac271ec9f4c5430e426f2e62b4382147868cf

henrikbrixandersen, 2021-01-12:
@Nukersson If your board does not have a storage partition, you can use `prj_no_storage.conf` instead.

Nukersson, 2021-01-14:
> -DCONF_FILE=prj_no_storage.conf

Thank you very much! It helped really.

henrikbrixandersen, 2021-01-14:
Can we close this issue?

Nukersson, 2021-01-14:
> Can we close this issue?

I've not found your advice in the documentation for mentioned sample. Therefore I've created new PR and linked to this bug. Can you review it please?

closed 2021-01-18 by carlescufi