zephyrproject-rtos/zephyr #31280: devicetree: Add a macro to easelly get optionnal devicetree GPIO properties

kurddt, 2021-01-13:
The idea is too allow easy handling of devices with optional GPIO properties. 
With a devicetree binding like the following:
```
   tx-ready-gpios:
    type: phandle-array
    required: false
```

It would be great to be able something like the following in the device driver configuration
```
const struct device_driver_config config_n = {
    .tx_ready_gpio_name = DT_GPIO_LABEL_OR(DT_DRV_INST(n), tx_ready_gpios),
    .tx_ready_pin = DT_GPIO_PIN_OR(DT_DRV_INST(n), tx_ready_gpios),
    .tx_ready_flags = DT_GPIO_FLAGS_OR(DT_DRV_INST(n), tx_ready_gpios),
};
```

Currently it's possible but cumbersome to achieve the same using  a combination of `NODE_HAS_PROPERTY` and `UTIL_AND ` or direct use of `DT_PROP_OR`

Example:
https://github.com/zephyrproject-rtos/zephyr/blob/43187e2723cfa6d095968f74726b24d7297329d5/drivers/eeprom/eeprom_at2x.c#L644

galak, 2021-01-13:
The implication is that the GPIO information would not than be described in devicetree.  This is not something I think we should support.  Why wouldn't you just describe the GPIO in the dts?

kurddt, 2021-01-13:
The pin will still be used to describe the peripheral doing the following

```	node: node@42 {
		compatible = "vendor,device";
		status = "okay";
		tx-ready-gpios = <&gpio0 7 0>;
	};
```

mbolivar-nordic, 2021-01-13:
I think I would rather see something like

```
struct gpio_dt_spec {
	const struct device *gpio;
	gpio_pin_t pin;
	gpio_dt_flags_t flags;
};

#define DT_GPIO_SPEC(node_id, gpio_pha) \
	COND_CODE_1(DT_NODE_HAS_PROP(node_id, gpio_pha),		\
		    (&(struct gpio_dt_spec) {				\
			    .gpio = DEVICE_DT_GET(			\
				    DT_GPIO_CTLR(node_id,		\
						 gpio_pha)),		\
			    .pin = DT_GPIO_PIN(node_id, gpio_pha),	\
			    .flags = DT_GPIO_FLAGS(node_id, gpio_pha),	\
		    }),							\
		    (NULL))
```

henrikbrixandersen, 2021-01-13:
> I think I would rather see something like

Yes, please. And similar for PWM and perhaps others.

kurddt, 2021-01-14:
I really like the idea too. 

One think though, in what you propose the optional behavior is mandatory: Unlike most of the DT macro there won't be a compilation error if gpio_pha does not exist. Maybe it would be good to have two variants: `DT_GPIO_SPEC` and `DT_GPIO_SPEC_OR_NULL` ?

mbolivar-nordic, 2021-01-14:
@kurddt sure, that seems reasonable.

pabigot, 2021-01-19:
> ```
> struct gpio_dt_spec {
> };

Yes, I've wanted this for a very long time too, and now that we can do build-time resolution of GPIO devices it becomes even more useful.

An issue is `arm,mps2-fpgaio-gpio` which has only one cell in the specifier.  That could (and perhaps should) be fixed, though.

mbolivar-nordic, 2021-01-19:
> An issue is `arm,mps2-fpgaio-gpio` which has only one cell in the specifier. That could (and perhaps should) be fixed, though.

Looking at the binding, I don't see an issue because there's a `pin` cell in the binding, and [DT_GPIO_FLAGS](https://docs.zephyrproject.org/latest/reference/devicetree/api.html#c.DT_GPIO_FLAGS) returns "the flags cell value [...], or zero if there is none".

pabigot, 2021-01-19:
True.  I would still say that flags like `GPIO_ACTIVE_LOW`, which is implemented at the wrapper level of the GPIO driver subsystem, should still be supported by this particular device, even if the hardware has nothing that can be controlled.  That, however, has nothing to do with the capability you're proposing.

mbolivar-nordic, 2021-01-19:
It's a shame we didn't have such a thing when `struct spi_cs_control` was being defined. With hindsight, something like this would have been a cleaner definition:

```c
struct spi_cs_control {
	uint32_t delay;
	struct gpio_dt_spec cs_pin;
};
```

Then the `SPI_CS_CONTROL_PTR_DT()` macro in #30536 could be built on top of what we're calling `DT_GPIO_SPEC()`

Probably too late to change at this point; it doesn't feel worth a stable API change.

At any rate, I'll put together an RFC so we can see how this plays out.

mbolivar-nordic, 2021-03-02:
>  I'll put together an RFC

Now done. I posted as a real PR rather than an RFC. https://github.com/zephyrproject-rtos/zephyr/pull/32805

mnkp, 2021-03-03:
That's a nice proposal, very useful as is. I'd like just briefly look into the future. The `gpios` property is of type `phandle-array` and as such can store several entries. It's possible that we will get another request to support this more complex case. 

If that happens we could either introduce `GPIO_DT_SPEC_GET_BY_IDX` macro or maybe make `GPIO_DT_SPEC_GET` return an array of `struct gpio_dt_spec`?

kurddt, 2021-03-08:
I don't this this one should be closed since as far as I understood it #32805 does not allow to have conditional/optional properties

kurddt, 2021-03-18:
So my preference would be to have both conditional and unconditional function available.

For the conditional one the else branch, I would prefer that the user provides the default value himself. it could look something like:

```
#define GPIO_DT_SPEC_GET_OR(node_id, prop, ...) COND_CODE_1(DT_NODE_HAS_PROP(node_id, prop), (GPIO_DT_SPEC_GET(node_id, prop)), __VA_ARGS__)

struct gpio_dt_spec button_a =
	GPIO_DT_SPEC_GET_OR(DT_NODELABEL(buttona), gpios, {.gpio = NULL, .pin = 42, .flag = 34});

```

Even if this this not really pretty (`{.gpio = NULL, .pin = 42, .flag = 34}` will be decomposed as 3 arguments)  I think it could work. The other solutions would be in order of preferences:
* always using {.gpio = NULL} as else branch like you suggested
* having 3 separate DT_GPIO_{LABEL,PIN,FLAGS}_OR like in my original proposal. So if one want to have optional gpios using `GPIO_DT_SPEC_GET` will not be possible

mbolivar-nordic, 2021-04-27:
> #define GPIO_DT_SPEC_GET_OR(node_id, prop, ...)

@kurddt I implemented this as prep work in https://github.com/zephyrproject-rtos/zephyr/pull/34589

kurddt, 2021-04-28:
Nice. I hope I will be able some time to update my drivers to use this as well as `DEVICE_DT_GET` and contribute them

PR #32805 (2021-03-06): gpio: add struct gpio_dt_spec
See #31280 for discussion and motivation.

This adds a structure and helper macro for converting a `foo-gpios` style devicetree property to a GPIO device and specifier data at build time.

Use the mesh_demo sample as a first user.

Fixes: #31280
commit d36c6a99 (2021-03-06): drivers: gpio: add struct gpio_dt_spec and helpers

This structure contains a GPIO device, pin number, and devicetree
flags. Add helpers for getting one out of the devicetree and doing
useful initialization tasks with it.

Fixes: #31280
PR #34589 (2021-05-05): samples: more DT cleanups
More incremental progress towards:

- dropping outmoded APIs like device_get_binding
- avoiding (driver-specific) DT_INST API usage in samples

As prep work:

Fixes: #31280
closed 2021-05-05 by carlescufi