zephyrproject-rtos/zephyr #31281: C++ support in Zephyr

alexanderwachter, 2021-01-13:
# This is a tracking issue for C++ enhancements for Zephyr
Add any missing features, things that work and things that don't work here.

## Missing features
List of missing features. The order defines the priority.
* Static local initialization. Static objects need to be guarded to get initialized properly (__cxa_guard_* functions). PR exists: #19754
* syscalls for class methods
* std::mutex
* std::atomic
* std::thread

## Things that don't work
* native_posix board behaves strange and does not terminate when compiled with gcov
* static objects (see missing feature)
* C++ standard library without exceptions. Exceptions getting enabled when the project links to the lib. Even with `--fno-exceptions`
## Things that work
* Use C++ for App development with restrictions
* STL
* Exceptions

## Nice to have

* exceptions with low footprint

nashif, 2021-01-13:
@pabigot FYI

pabigot, 2021-01-14:
@alexanderwachter could you perhaps update #18554 instead, and change its description to reflect "status as of 2.5"?  We don't need multiple things tracking the gaps.

abrodkin, 2021-01-18:
@evgeniy-paltsev @ruuddw FYI