zephyrproject-rtos/zephyr #31289: Problems building grub2 bootloader for Zephyr

jaslob, 2021-01-13:
I want to use Zephyr as a User VM in ACRN. Therefore I followed the tutorials in https://projectacrn.github.io/latest/tutorials/using_zephyr_as_uos.html 
I followed the zephyr getting started guide on my Ubuntu and when I want to build the grub2 bootloader image in step 2 I get the following message:

jas@ubuntu:~/zephyrproject/zephyr$ ./boards/x86/common/scripts/build_grub.sh x86_64
~/zephyrproject/zephyr/boards/x86/common/scripts/grub ~/zephyrproject/zephyr
~/zephyrproject/zephyr/boards/x86/common/scripts/grub/src ~/zephyrproject/zephyr/boards/x86/common/scripts/grub ~/zephyrproject/zephyr
HEAD is now at 2a2e10c1b Release 2.04
Removing INSTALL.grub
Removing build-aux/
~/zephyrproject/zephyr/boards/x86/common/scripts/grub ~/zephyrproject/zephyr
~/zephyrproject/zephyr/boards/x86/common/scripts/grub/src ~/zephyrproject/zephyr/boards/x86/common/scripts/grub ~/zephyrproject/zephyr
./bootstrap: 470: autoconf: not found
./bootstrap: Error: 'autoconf' not found
./bootstrap: 470: automake: not found
./bootstrap: Error: 'automake' not found

./bootstrap: Please install the prerequisite programs
Gnulib not yet bootstrapped; run ./bootstrap instead.
./boards/x86/common/scripts/build_grub.sh: line 37: ./configure: No such file or directory
make: *** No targets specified and no makefile found.  Stop.
./boards/x86/common/scripts/build_grub.sh: line 41: ./grub-mkimage: No such file or directory
~/zephyrproject/zephyr/boards/x86/common/scripts/grub ~/zephyrproject/zephyr
cp: cannot stat 'src/grub_x86_64.efi': No such file or directory

I'd be thankful for help.

dcpleung, 2021-01-14:
Do you have `autoconf` and `automake` installed? IIRC, in Ubuntu, the packages are of the same names.

jaslob, 2021-01-28:
Thanks. It took some time but it did finally work.

closed 2021-01-28 by jaslob

klogg, 2023-09-27:
> Thanks. It took some time but it did finally work.

hey @jaslob can you still run grub on x86 with Zephyr? somehow it does not work for us, can you share how it works for you?

dcpleung, 2023-09-27:
What were the error messages?