zephyrproject-rtos/zephyr #31299: tests/kernel/mbox/mbox_usage failed on hsdk board

IRISZZW, 2021-01-14:
**Describe the bug**
tests/kernel/mbox/mbox_usage failed on hsdk board

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b hsdk -d build -p auto tests\kernel\mbox\mbox_usage
2. west flash -r mdb-hw
3. See error

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v2.4.0-3009-gfc1b5de4c307  ***
Running test suite test_mbox
===================================================================
START - test_msg_receiver
E: ***** Exception vector: 0x1, cause code: 0x0, parameter 0x0
 PASS - test_msg_receiver
E: Address 0xfffffffc
===================================================================
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
START - test_msg_receiver_unlimited
E: Current thread: 0x9000791c (unknown)
Sender UNLIMITED
E: Halting system
```

**Environment (please complete the following information):**
 - OS: (Linux)
 - Toolchain (Zephyr SDK)
 - master branch (zephyr-v2.4.0-3009-gfc1b5de4c307)

nashif, 2021-02-09:
@evgeniy-paltsev  @abrodkin  can you please confirm if this still fails?

IRISZZW, 2021-04-15:
I have checked our daily CI, this test can run successful for a long time.
One day, it’s like a miracle, it disappeared..

```
*** Booting Zephyr OS build zephyr-v2.5.0-2257-ge4a349ff76e3  ***
Running test suite test_mbox
===================================================================
START - test_msg_receiver
 PASS - test_msg_receiver in 0.1 seconds
===================================================================
START - test_msg_receiver_unlimited
Sender UNLIMITED
 PASS - test_msg_receiver_unlimited in 0.2 seconds
===================================================================
START - test_multi_thread_send_get
 PASS - test_multi_thread_send_get in 0.32 seconds
===================================================================
Test suite test_mbox succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL 
```

@abrodkin  May we closed this issue?

abrodkin, 2021-04-15:
Sure, let's close this one. If we start seeing it agin we'll re-open it.

closed 2021-04-15 by abrodkin