zephyrproject-rtos/zephyr #31301: intel_adsp_cavs15: run testing with twister failed.

KangJianX, 2021-01-14:
Describe the bug
Use twister to run kernel testcases failed

To Reproduce
Steps to reproduce the behavior:
 twister -p intel_adsp_cavs15 --device-testing -T tests/kernel/mutex/mutex_api/ --west-flash="/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/flash.sh,/home/ztest/work/sof/rimage/keys/otc_private_key.pem" --device-serial-pty="/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/adsplog.py,--no-history"

see error:
INFO    - Zephyr version: zephyr-v2.4.0-2966-gfa6ac271ec9f
INFO    - JOBS: 4
INFO    - Building initial testcase list...
INFO    - 1 test suites (1 configurations) selected, 0 configurations discarded due to filters.

Device testing on:

| Platform          | ID   | Serial device   |
|-------------------|------|-----------------|
| intel_adsp_cavs15 |      |                 |

INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue

**ERROR   - intel_adsp_cavs15         tests/kernel/mutex/mutex_api/kernel.mutex          FAILED: Timeout
ERROR   - see: /home/ztest/work/zephyrproject/zephyr/twister-out/intel_adsp_cavs15/tests/kernel/mutex/mutex_api/kernel.mutex/device.log**
INFO    - Total complete:    1/   1  100%  skipped:    0, failed:    1
INFO    - 0 of 1 test configurations passed (0.00%), 1 failed, 0 skipped with 0 warnings in 95.31 seconds
INFO    - In total 7 test cases were executed, 0 skipped on 1 out of total 325 platforms (0.31%)
INFO    - 1 test configurations executed on platforms, 0 test configurations were only built.

Hardware distribution summary:

| Board             | ID   |   Counter |
|-------------------|------|-----------|
| intel_adsp_cavs15 |      |         1 |
INFO    - Saving reports...
INFO    - Writing xunit report /home/ztest/work/zephyrproject/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /home/ztest/work/zephyrproject/zephyr/twister-out/twister_report.xml...
INFO    - Writing JSON report /home/ztest/work/zephyrproject/zephyr/twister-out/twister.json
INFO    - Run completed

**The device.log:**
Traceback (most recent call last):
  File "/home/ztest/.local/bin/west", line 11, in <module>
    sys.exit(main())
  File "/home/ztest/.local/lib/python3.6/site-packages/west/app/main.py", line 768, in main
    app.run(argv or sys.argv[1:])
  File "/home/ztest/.local/lib/python3.6/site-packages/west/app/main.py", line 106, in run
    self.run_command(argv)
  File "/home/ztest/.local/lib/python3.6/site-packages/west/app/main.py", line 334, in run_command
    self.run_extension(args.command, argv)
  File "/home/ztest/.local/lib/python3.6/site-packages/west/app/main.py", line 397, in run_extension
    command.run(args, unknown, self.topdir, manifest=self.manifest)
  File "/home/ztest/.local/lib/python3.6/site-packages/west/commands.py", line 116, in run
    self.do_run(args, unknown)
  File "/home/ztest/work/zephyrproject/zephyr/scripts/west_commands/sign.py", line 185, in do_run
    signer.sign(self, build_dir, bcfg, formats)
  File "/home/ztest/work/zephyrproject/zephyr/scripts/west_commands/sign.py", line 434, in sign
    rimage_conf = pathlib.Path(cache['RIMAGE_CONFIG_PATH'])
  File "/home/ztest/work/zephyrproject/zephyr/scripts/west_commands/zcmake.py", line 255, in __getitem__
    return self._entries[name].value
KeyError: 'RIMAGE_CONFIG_PATH'

Environment (please complete the following information):
OS: Fedora28
Toolchain: Zephyr-sdk-0.12.0
Commit ID: fa6ac271ec9

KangJianX, 2021-01-14:
@andyross Hi Andy, I have updated the rimage in zephyrproject/modules/audio/sof/rimage and install it. The testing still failed.

KangJianX, 2021-01-14:
The latest commit build failed too. 
**Log:**
[10/151] Generating include/generated/driver-validation.h
[11/151] Building C object zephyr/CMakeFiles/offsets.dir/arch/xtensa/core/offsets/offsets.c.obj
FAILED: zephyr/CMakeFiles/offsets.dir/arch/xtensa/core/offsets/offsets.c.obj 
ccache /opt/sdk/zephyr-sdk-0.11.4/xtensa/intel_apl_adsp/xtensa-zephyr-elf/bin/xtensa-zephyr-elf-gcc -DBUILD_VERSION=zephyr-v2.4.0-3010-g7d8b31f2da76 -DKERNEL -D_FORTIFY_SOURCE=2 -D__ZEPHYR__=1 -I../../../../../../../kernel/include -I../../../../../../../arch/xtensa/include -I../../../../../../../include -Izephyr/include/generated -I../../../../../../../soc/xtensa/intel_adsp/cavs_v15 -I../../../../../../../soc/xtensa/intel_adsp/cavs_v15/include -I../../../../../../../soc/xtensa/intel_adsp/common/include -I../../../../../../../subsys/testsuite/include -I../../../../../../../subsys/testsuite/ztest/include -I/var/lib/buildkite-agent/zephyr-module-cache/modules/hal/xtensa/include -I/var/lib/buildkite-agent/zephyr-module-cache/modules/hal/xtensa/zephyr/soc/intel_apl_adsp -isystem ../../../../../../../lib/libc/minimal/include -isystem /opt/sdk/zephyr-sdk-0.11.4/xtensa/intel_apl_adsp/xtensa-zephyr-elf/bin/../lib/gcc/xtensa-zephyr-elf/9.2.0/include -isystem /opt/sdk/zephyr-sdk-0.11.4/xtensa/intel_apl_adsp/xtensa-zephyr-elf/bin/../lib/gcc/xtensa-zephyr-elf/9.2.0/include-fixed -Os -imacros /workdir/zephyr/twister-out/intel_adsp_cavs15/tests/kernel/semaphore/semaphore_error_cases/kernel.semaphore/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -imacros /workdir/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wno-address-of-packed-member -Wno-unused-but-set-variable -Werror=implicit-int -Werror -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/workdir/zephyr/tests/kernel/semaphore/semaphore_error_cases=CMAKE_SOURCE_DIR -fmacro-prefix-map=/workdir/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/workdir=WEST_TOPDIR -ffunction-sections -fdata-sections -mlongcalls -std=c99 -nostdinc -MD -MT zephyr/CMakeFiles/offsets.dir/arch/xtensa/core/offsets/offsets.c.obj -MF zephyr/CMakeFiles/offsets.dir/arch/xtensa/core/offsets/offsets.c.obj.d -o zephyr/CMakeFiles/offsets.dir/arch/xtensa/core/offsets/offsets.c.obj   -c /workdir/zephyr/arch/xtensa/core/offsets/offsets.c
In file included from ../../../../../../../include/arch/xtensa/xtensa_api.h:10,
                 from ../../../../../../../include/arch/xtensa/irq.h:9,
                 from ../../../../../../../include/arch/xtensa/arch.h:27,
                 from ../../../../../../../include/arch/cpu.h:27,
                 from ../../../../../../../include/kernel_includes.h:33,
                 from ../../../../../../../include/kernel.h:17,
                 from /workdir/zephyr/arch/xtensa/core/offsets/offsets.c:26:
/var/lib/buildkite-agent/zephyr-module-cache/modules/hal/xtensa/include/xtensa/core-macros.h:141:11: fatal error: xtensa/tie/xt_datacache.h: No such file or directory
  141 | # include 
      |           ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.

KangJianX, 2021-01-20:
This issue not be fixed, it still can be reproduce.

KangJianX, 2021-01-20:
The twister always timeout, because west can not get RIMAGE_CONFIG_PATH **File "/home/ztest/work/zephyrproject/zephyr/scripts/west_commands/sign.py", line 434, in sign
rimage_conf = pathlib.Path(cache['RIMAGE_CONFIG_PATH'])** , this error is in device.log of twister-out. **KeyError: 'RIMAGE_CONFIG_PATH'** .
And rimage can be found in /usr/local/bin/

lyakh, 2021-01-25:
twister is building test applications for different platforms. In this case it's trying to build (and run) a kernel mutex test application. Which means it isn't building SOF at all. And the `RIMAGE_CONFIG_PATH` variable is cached as a part of the SOF building process. So twister isn't finding it. I'm not sure what the best way to fix the problem would be. If those test applications are indeed supposed to run on the hardware, they have to be signed. For signing rimage needs a toml configuration file and it's located within the modules/audio/sof path. And if you don't build SOF you "don't have access" to that path because we don't want to reach out to files across modules. Maybe the correct solution would be moving rimage to scripts/west_commands/ but rimage is an external project with a sub-project, so, maybe that isn't something that Zephyr will want to do.

PR #31313 (2021-01-14): xtensa: remove core-macros.h from xtensa HAL
core-macros.h includes other files not part of the xtensa HAL, make this
esp32 specific

Fixes #31301
commit 232851a4 (2021-01-14): xtensa: remove core-macros.h from xtensa HAL

core-macros.h includes other files not part of the xtensa HAL, make this
esp32 specific

Fixes #31301
PR #31574 (2021-01-26): sof: update to latest SOF master
Sync with SOF tree 385cf89c7

Fixes #31301
commit e5f054b0 (2021-01-26): sof: update to latest SOF master

Sync with SOF tree 385cf89c7

Fixes #31301
closed 2021-01-26 by nashif