zephyrproject-rtos/zephyr #31308: hci_usb error with status 0x11 (Invalid Index)

furkanhtk, 2021-01-14:
**Describe the bug**
nRF5340 DK
I tried hci_uart and hci_usb samples but it did not work with bluez,bluethootctl and hciconfig. 
I looked at these Issues.
@https://github.com/zephyrproject-rtos/zephyr/issues/3230
@https://github.com/zephyrproject-rtos/zephyr/issues/11016

**I got this error when I try to set static-addr to nRF5340 DK:**
```
[hci1]# static-addr 00:00:00:00:00:11
Set static address failed with status 0x11 (Invalid Index)
```
Any suggestion?

prj.conf
```
CONFIG_STDOUT_CONSOLE=y
CONFIG_GPIO=y
CONFIG_SERIAL=y
CONFIG_UART_INTERRUPT_DRIVEN=y

CONFIG_BT=y
CONFIG_BT_HCI=y
CONFIG_BT_CTLR=y

CONFIG_USB=y
CONFIG_USB_DEVICE_STACK=y
CONFIG_USB_DEVICE_BLUETOOTH=y
CONFIG_USB_DEVICE_BLUETOOTH_VS_H4=y

CONFIG_USB_DEVICE_VID=0x167f
CONFIG_USB_DEVICE_PID=0x3001

CONFIG_USB_DEVICE_MANUFACTURER="Company"
CONFIG_USB_DEVICE_PRODUCT="Furkan"
CONFIG_USB_DEVICE_SN="123abc"
```

**Build the hci_usb**
1. west build -p auto -b nrf5340dk_nrf5340_cpuapp samples/bluetooth/hci_usb
2. west flash

```
lsusb 
Bus 003 Device 009: ID 167f:3001
```

```
lsusb -vd ID 167f:3001
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x167f
  idProduct          0x3001
  bcdDevice            2.04
  iManufacturer           1 Company
  iProduct                2 Furkan
  iSerial                 3 B1F02D
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           39
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       224 Wireless
      bInterfaceSubClass      1 Radio Frequency
      bInterfaceProtocol      1 Bluetooth
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered
```
```
hciconfig 
hci1:   Type: Primary  Bus: USB
        BD Address: 00:00:00:00:00:00  ACL MTU: 0:0  SCO MTU: 0:0
        DOWN
        RX bytes:0 acl:0 sco:0 events:0 errors:0
        TX bytes:3 acl:0 sco:0 commands:1 errors:0
```
```
root@xxx:~# btmgmt
[mgmt]# select 1
Selected index 1
[hci1]# static-addr 00:00:00:00:00:11
Set static address failed with status 0x11 (Invalid Index)
```
```
root@smarcimx8m4g:~# hciconfig hci1 up
Can't init device hci1: Connection timed out (110)
```

**Environment:**
 - OS: Build in Linux Ubuntu WM
 - Toolchain  Zephyr SDK
- `Linux smarcimx8m4g 4.14.98-gab14c2a-dirty #1 SMP PREEMPT Tue Sep 8 22:39:28 +03 2020 aarch64 aarch64 aarch64 GNU/Linux
`

furkanhtk, 2021-01-14:
@carlescufi @jfischer-no @dcpleung

jfischer-no, 2021-01-14:
>Linux smarcimx8m4g 4.14.98-gab14c2a-dirty #1 SMP PREEMPT Tue Sep 8 22:39:28 +03 2020 aarch64 aarch64 aarch64 GNU/Linux 

I guess the answer from #11016 explains it [quite well](https://github.com/zephyrproject-rtos/zephyr/issues/11016#issuecomment-435925274). Your kernel or BlueZ are a bit outdated.

carlescufi, 2021-02-01:
> > Linux smarcimx8m4g 4.14.98-gab14c2a-dirty #1 SMP PREEMPT Tue Sep 8 22:39:28 +03 2020 aarch64 aarch64 aarch64 GNU/Linux
> 
> I guess the answer from #11016 explains it [quite well](https://github.com/zephyrproject-rtos/zephyr/issues/11016#issuecomment-435925274). Your kernel or BlueZ are a bit outdated.

Yep, this seems like the same issue. Closing, please reopen if this has not solved it.

closed 2021-02-01 by carlescufi

furkanhtk, 2021-02-10:
Hi,
As you suggested, I tried it on Raspberry with new kernel and Bluez versions and encountered the same error.
`Linux raspberrypi 5.4.83-v7+ #1379 SMP Mon Dec 14 13:08:57 GMT 2020 arm7l GsNU/Linux`
with bluez version : `5.50-1.2~deb10u1+rpt2`
Thanks for help.
@carlescufi