zephyrproject-rtos/zephyr #31323: subsys/sntp

WilliamGFish, 2021-01-14:
**Describe the bug**
Compilation warning regards the SNTP subsys.

/subsys/net/lib/sntp/sntp.c:161:3: warning: implicit declaration of function 'close' [-Wimplicit-function-declaration]
  161 |   close(ctx->sock.fd);

**Expected behavior**
No compilation warning as the "close" function is available.

**Impact**
Annoyance, does not appear to have any impact on functionality

**Logs and console output**
/subsys/net/lib/sntp/sntp.c:161:3: warning: implicit declaration of function 'close' [-Wimplicit-function-declaration]
  161 |   close(ctx->sock.fd);

**Environment (please complete the following information):**
 -  Windows 10
 - West latest and GCC
 - 2.4.99

**Additional context**
Add any other context about the problem here.

pfalcon, 2021-02-04:
@WilliamGFish: Can you please update the subject of this ticket to summarize its content? Please see https://github.com/zephyrproject-rtos/zephyr/issues for examples of good ticket subjects.

nashif, 2021-02-09:
Can you provide some more context and what test/sample on which platform you are getting this?

closed 2021-04-25 by github-actions[bot]