zephyrproject-rtos/zephyr #31327: ci compliance failures due to intel_adsp_cavs25 sample

pabigot, 2021-01-14:
samples/audio/sof/boards/intel_adsp_cavs25.conf references a bunch of Kconfig symbols that are out-of-tree, resulting in CI failures as with https://github.com/zephyrproject-rtos/zephyr/pull/28977/checks?check_run_id=1704198353

pabigot, 2021-01-14:
This went away when I rebased my branch on current master in https://github.com/zephyrproject-rtos/zephyr/pull/28977/checks?check_run_id=1704273529  so these Kconfig variables must be defined in an external module; but this suggests the compliance tests aren't properly setting up the environment for what the PR looks like when rebased on master.

nashif, 2021-01-18:
compliance tests were not setting up west and modules, now they are, any PRs that were submitted prior to that change will fail with the above, so those PRs will need to be rebased.

pabigot, 2021-01-18:
Aren't all the compliance tests including CI done on the branch after it's been rebased on current master?  If not, it's not giving us a correct view of what will happen if it's merged.

PR #31929 (2021-02-03): actions: compliance: Rebase before running `west update`
Instead of rebasing just before running the compliance script, do so
before running `west update`, so that the main manifest, west.yml, is up
to date with the latest master before west inspects it.

Fixes #31327.
commit 777c5c99 (2021-02-03): actions: compliance: Rebase before running `west update`

Instead of rebasing just before running the compliance script, do so
before running `west update`, so that the main manifest, west.yml, is up
to date with the latest master before west inspects it.

Fixes #31327.
closed 2021-02-03 by nashif