zephyrproject-rtos/zephyr #31333: adding a periodic k_timer causes k_msleep to never return in tests/kernel/context

andrewboie, 2021-01-14:
**Describe the bug**
I was trying to figure out why my demand paging PR was failing tests/kernel/context and was able to isolate the issue to a periodic timer hook installed by my eviction algorithm at boot.

I was then able to come up with a trivial scenario which reproduces this in the master branch.
Something seems to be very broken with timeouts.

**To Reproduce**
Apply this patch:

```
diff --git a/tests/kernel/context/src/main.c b/tests/kernel/context/src/main.c
index a0e4318fa9..a2b10e492b 100644
--- a/tests/kernel/context/src/main.c
+++ b/tests/kernel/context/src/main.c
@@ -1154,6 +1154,12 @@ void test_kernel_thread(void)
 
 }
 
+static void timer_hook(struct k_timer *timer)
+{
+       printk("my timer ran\n");
+}
+static K_TIMER_DEFINE(my_timer, timer_hook, NULL);
+
 /*test case main entry*/
 void test_main(void)
 {
@@ -1161,6 +1167,9 @@ void test_main(void)
 
        kernel_init_objects();
 
+       k_timer_start(&my_timer, K_NO_WAIT,
+                     K_MSEC(100));
+
        ztest_test_suite(context,
                         ztest_unit_test(test_kernel_interrupts),
                         ztest_1cpu_unit_test(test_kernel_timer_interrupts),
```

And then run tests/kernel/context on qemu_x86:

```
[0/1] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: qemu32,+nx,+pae
SeaBIOS (version rel-1.12.1-0-ga5cab58-dirty-20200625_115407-9426dddc0a1f-zephyr
)
Booting from ROM..*** Booting Zephyr OS build zephyr-v2.4.0-3052-gf02454f68a02  ***
Running test suite context
===================================================================
START - test_kernel_interrupts
 PASS - test_kernel_interrupts
===================================================================
START - test_kernel_timer_interrupts
my timer ran
 PASS - test_kernel_timer_interrupts
===================================================================
START - test_kernel_ctx_thread
Testing k_current_get() from an ISR and thread
Testing k_is_in_isr() from an ISR
Testing k_is_in_isr() from a preemptible thread
 PASS - test_kernel_ctx_thread
===================================================================
START - test_busy_wait
Thread busy waiting for 20000 usecs
Thread busy waiting completed
Thread busy waiting for 20000 usecs (irqs locked)
Thread busy waiting completed (irqs locked)
 PASS - test_busy_wait
===================================================================
START - test_k_sleep
 thread sleeping for 50 milliseconds
qemu-system-i386: warning: icount sleep disabled and no active timers
```

The timer fires once (note the printout), and then the system deadlocks when `test_k_sleep` calls `k_msleep()`.

**Expected behavior**
The periodic timer to keep printing, and the test to complete successfully.

I am a little puzzled that the initial printout of "my timer ran" takes place much later than the call site of k_timer_start(). Maybe I am misunderstanding the (rather terse) documentation for this API.

**Impact**
I do not believe I am mis-using the k_timer APIs.
I don't think there is something special about the test.
I fear this is a kernel bug, but this is the only test that fails, and there are many other tests that sleep.

andyross, 2021-01-15:
FWIW: there is something special about this test.

That timer interrupt you requested arrives while during the test_kernel_timer_interrupts case, while the timer interrupt is about to be or has just been disabled for a long time while the test spins.  And it's asking for a repeating timer, so it's trying to reschedule a new event at a time where the timer interrupt count is way, way off.  Presumably we're hitting an edge case in the handling for very delayed interrupts or something similar?  Will investigate.

andyross, 2021-01-15:
This actually looks like an arch bug, I think.  What seems to be happening is that the test calls `arch_irq_disable()` on the timer interrupt.  Then it spins for a long time, and during that period the moment where the timer driver was supposed to have fired its next interrupt arrive, but of course the interrupt was masked.  Then the test calls `arch_irq_enable()` to re-enable the interrupt.  But **on x86, at least, the now-unmasked interrupt does not fire**.

Then the test goes on, and the next time the timeout system is asked to do something it trips over the fact that it's in an impossible state, where a timeout exists in the list that should have been fired in the past, and some negative numbers result and everything blows up.

But the root cause seems to be that arch_irq_enable() should (per the test expectations) be acting like arch_irq_unlock() and merely unmasking flagged interrupts.  But that's not what it's doing.  And we don't seem to have ever thought about the distinction (at least I haven't), nor documented the rule.

(Interestingly one gotcha is that the HPET driver is currently asking for an EDGE triggered interrupt per its DT configuration, which would have this effect.  But hard coding that doesn't seem to have fixed it and APIC behavior is always more complicated than it seems.)

andyross, 2021-01-15:
This minimal rig shows the issue fairly clearly.  You can hack numbers into a spin loop to adjust the delay (i.e. without involving the timer driver at all), and prove that the timer interrupt fails to arrive if the event happens while the interrupt is disabled:

```c
#include <zephyr.h>
#include <timeout_q.h>

/* Hard-coded to HPET/x86! */
#define TIMER_IRQ 2

BUILD_ASSERT(CONFIG_HPET_TIMER);

static void timeout_fn(struct _timeout *t)
{
	printk("%s\n", __func__);
}

void main(void)
{
	static struct _timeout to = {};

	z_add_timeout(&to, timeout_fn, K_TICKS(1));

	/* Note: when delay is small (<100k or so on my box) and the
	 * loop completes before the event, the timer interrupt fires.
	 * When it is too long, the now-unmasked interrupt gets lost.
	 */
	printk("spin with IRQ disabled...\n");
	arch_irq_disable(TIMER_IRQ);
	for(volatile int i=0; i<100000; i++);
	arch_irq_enable(TIMER_IRQ);
	printk("IRQ re-enabled\n");

	while(1);
}
```

andyross, 2021-01-15:
I need to give this a little thought as to how we want to handle it.  The IRQ en/disable API is a deep and weird API.  It's not really intended for locking, so there's no particular reason for it to demand that the pending state be preserved I guess.  But if it's not for locking, then what is it good for anyway?  My guess is most platforms probably have some variant of this bug, so we can't just fix it with documentation.

Also it's probably possible to harden the timeout API against this case, which would fix the proximate cause.  There's no way we can hook irq_enable() to force a timer interrupt, but we could at least detect the overflow internally and recover at some point in the future when someone calls into a timeout function.

PR #31497 (2021-01-21): arch: tests: Document interrupt delivery behavior after arch_irq_disa…
…ble()

Upcoming changes from Andrew that add a global timeout to the kernel
broke because of some voodoo behavior in the kernel/context test.  It
will use arch_irq_disable() on the timer interrupt directly to prevent
interrupts and measure timekeeping in their absence.  But some
architectures[1] don't reliably deliver interrupts that arrive, which
means that a running timeout that exists across this period will
result in a corrupt timeout queue.

Document that rule for architectures, move the offending test to the
end of t [... cut]
commit e932a153 (2021-01-21): arch: tests: Document interrupt delivery behavior after arch_irq_disable()

Upcoming changes from Andrew that add a global timeout to the kernel
broke because of some voodoo behavior in the kernel/context test.  It
will use arch_irq_disable() on the timer interrupt directly to prevent
interrupts and measure timekeeping in their absence.  But some
architectures[1] don't reliably deliver interrupts that arrive, which
means that a running timeout that exists across this period will
result in a corrupt timeout queue.

Document that rule for architectures, move the offending test to the
end of the  [... cut]
closed 2021-01-21 by nashif