zephyrproject-rtos/zephyr #31338: mimxrt1050_evk: running tests/kernel/fpu_sharing/float_disable/ failed

Zhaoningx, 2021-01-15:
To Reproduce
Steps to reproduce the behavior:
twister -p mimxrt1050_evk --device-testing --device-serial /dev/ttyACM0 -T tests/kernel/fpu_sharing/float_disable/
see error

*** Booting Zephyr OS build zephyr-v2.4.0-3009-gfc1b5de4c307  ***
Running test suite suite_k_float_disable

===================================================================
START - test_k_float_disable_common
PASS - test_k_float_disable_common

===================================================================
SE: ***** MPU FAULT *****
E:   Stacking error (context area might be not valid)
E: r0/a1:  0x6000e8a5  r1/a2:  0x00000000  r2/a3:  0x6001049e
E: r3/a4:  0x00000000 r12/ip:  0x6000f2fc r14/lr:  0x00000000
E:  xpsr:  0x6000f200
E: s[ 0]:  0xffffffff  s[ 1]:  0x00000000  s[ 2]:  0xffffffff  s[ 3]:  0x00000000
E: s[ 4]:  0x00000000  s[ 5]:  0x00000000  s[ 6]:  0xffffffff  s[ 7]:  0xffffffff
E: s[ 8]:  0xffffffff  s[ 9]:  0x00000000  s[10]:  0xffffffff  s[11]:  0xffffffff
E: s[12]:  0xffffffff  s[13]:  0x00000000  s[14]:  0xffffffff  s[15]:  0xffffffff
E: fpscr:  0x6000f370
E: Faulting instruction address (r15/pc): 0x60010497
E: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
E: Current thread: 0x800004a0 (main)
E: Halting system

Environment (please complete the following information):

OS: Fedora33
Toolchain: zephyr-sdk-0.12.0
Commit ID: fc1b5de4c307

pabigot, 2021-01-19:
Please retest on current HEAD since #31173 has been merged.

MaureenHelm, 2021-01-26:
Fixed by #31173

closed 2021-01-26 by MaureenHelm