zephyrproject-rtos/zephyr #31339: nsim: running tests/ztest/error_hook/ failed

Zhaoningx, 2021-01-15:
To Reproduce
Steps to reproduce the behavior:
twister -p nsim_em -T tests/ztest/error_hook/
see error

*** Booting Zephyr OS build zephyr-v2.4.0-3009-gfc1b5de4c307  ***^M
Running test suite error_hook_tests^M

===================================================================^M
START - test_catch_assert_fail^M
ASSERTION FAIL [a != ((void *)0)] @ WEST_TOPDIR/zephyr/tests/ztest/error_hook/src/main.c:41^M
        parameter a should not be NULL!^M
Caught assert failed^M
Assert error expected as part of test case.^M
 PASS - test_catch_assert_fail^M

===================================================================^M
**START - test_catch_fatal_error^M
case type is 0^M
b is 11588^M
 FAIL - test_catch_fatal_error^M**

===================================================================^M
START - test_catch_assert_in_isr^M
ASSERTION FAIL [a != ((void *)0)] @ WEST_TOPDIR/zephyr/tests/ztest/error_hook/src/main.c:41^M
        parameter a should not be NULL!^M
Caught assert failed^M
Assert error expected as part of test case.^M
 PASS - test_catch_assert_in_isr^M

Environment (please complete the following information):

OS: Fedora33
Toolchain: zephyr-sdk-0.12.0
Commit ID: fc1b5de

enjiamai, 2021-01-15:
The assertion failed due to access a NULL address did not trigger an fatal error in nsim_em board. Let me see how to modify me test case to adapt that.
This root cause of this issue is the same with https://github.com/zephyrproject-rtos/zephyr/issues/31344

enjiamai, 2021-01-26:
An new issue also has the root cause of this:
https://github.com/zephyrproject-rtos/zephyr/issues/31419

enjiamai, 2021-02-02:
duplicate of #31419 ,  so close it.

PR #31773 (2021-02-03): tests: ztest: Fix issues due to test case of accessing NULL address
Fix issue #31339 and #31419, test case fail due to access NULL did not
trigger a fatal error in some platform such as nsim_em and iotdk.

Fixes #31339 
Fixes #31419
closed 2021-02-02 by enjiamai