zephyrproject-rtos/zephyr #31341: Bluetooth: Use CHECKIF for validating input in API functions

Thalley, 2021-01-15:
**Is your enhancement proposal related to a problem? Please describe.**
There was recently a discussion on Slack regarding how we should validate/verify arguments in API functions.

In internal functions we can usually use __ASSERT, but for API functions, where values are provided by users/applications, we probably want to add additional checks.
**Describe the solution you'd like**
The CHECKIF macro can be configured to either be an __ASSERT, a regular run-time check or even be ignored, giving bit more flexibility than an __ASSERT or a regular runtime check.

I suggest that for all public functions, we start using CHECKIF for all necessary parameters (i.e. NULL pointer checks, array indexes, etc.).

**Describe alternatives you've considered**
Could perhaps be expended for other subsystems besides Bluetooth. 

**Additional context**
N/A

jhedberg, 2024-12-19:
Comments from Bluetooth WG:

 - Documented API promises shouldn't depend on `CHECKIF()` (e.g. if the API promises to return a specific error when given a specific input value)
 - API documentation may need updating where it's currently making such promises.
 - Consider removing `CHECKIF()` from some existing places due to the above

Thalley, 2024-12-21:
Just a quick measurement using the LE Audio shell: 

Checks disabled (CONFIG_NO_RUNTIME_CHECKS)
```
Memory region         Used Size  Region Size  %age Used
           FLASH:      958764 B         1 MB     91.43%
             RAM:      404902 B       448 KB     88.26%
        IDT_LIST:          0 GB        32 KB      0.00%
```

Default (just if checks)
```
Memory region         Used Size  Region Size  %age Used
           FLASH:      990352 B         1 MB     94.45%
             RAM:      404902 B       448 KB     88.26%
        IDT_LIST:          0 GB        32 KB      0.00%

As asserts (CONFIG_ASSERT_ON_ERRORS)
```
Memory region         Used Size  Region Size  %age Used
           FLASH:      992368 B         1 MB     94.64%
             RAM:      404902 B       448 KB     88.26%
        IDT_LIST:          0 GB        32 KB      0.00%
```

So it's clear that the `if` checks are definitely not nothing (nearly 3% of the ROM), which is why `CHECKIF` was useful. 

Doing this test clearly also shows that we aren't testing it, as setting `CONFIG_NO_RUNTIME_CHECKS=y` causes quite a few build warnings.

Thalley, 2025-01-02:
An update to the above: I figured that log statements were probably skewing the results, so here it is without logs enabled:

Checks disabled (CONFIG_NO_RUNTIME_CHECKS)
```
Memory region         Used Size  Region Size  %age Used
           FLASH:      689568 B         1 MB     65.76%
             RAM:      400374 B       448 KB     87.27%
        IDT_LIST:          0 GB        32 KB      0.00%
```
        
Default (just if checks)
```
Memory region         Used Size  Region Size  %age Used
           FLASH:      694128 B         1 MB     66.20%
             RAM:      400374 B       448 KB     87.27%
        IDT_LIST:          0 GB        32 KB      0.00%
```