zephyrproject-rtos/zephyr #31344: iotdk: running tests/ztest/error_hook/ failed

Zhaoningx, 2021-01-15:
To Reproduce
Steps to reproduce the behavior:
twister -p iotdk --device-testing --device-serial /dev/ttyUSB0 -T tests/ztest/error_hook/
see error
*** Booting Zephyr OS build zephyr-v2.4.0-3009-gfc1b5de4c307  ***
Running test suite error_hook_tests

===================================================================
START - test_catch_assert_fail
ASSERTION FAIL [a != ((void *)0)] @ WEST_TOPDIR/zephyr/tests/ztest/error_hook/src/main.c:41
parameter a should not be NULL!
Caught assert failed
Assert error expected as part of test case.
PASS - test_catch_assert_fail

===================================================================
START - test_catch_fatal_error
case type is 0
b is 1024
FAIL - test_catch_fatal_error

===================================================================
START - test_catch_assert_in_isr
ASSERTION FAIL [a != ((void *)0)] @ WEST_TOPDIR/zephyr/tests/ztest/error_hook/src/main.c:41
parameter a should not be NULL!
Caught assert failed
Assert error expected as part of test case.
PASS - test_catch_assert_in_isr

===================================================================
START - test_catch_z_oops
PASS - test_catch_z_oops

===================================================================
Test suite error_hook_tests failed.

===================================================================
PROJECT EXECUTION FAILED

Environment (please complete the following information):

OS: Fedora33
Toolchain: zephyr-sdk-0.12.0
Commit ID: fc1b5de

enjiamai, 2021-01-15:
The assertion failed due to access a NULL address did not trigger an fatal error in nsim_em board. Let me see how to modify me test case to adapt that.
This root cause of this issue is the same with https://github.com/zephyrproject-rtos/zephyr/issues/31339

nashif, 2021-01-18:
duplicate of #31339

closed 2021-01-18 by nashif