zephyrproject-rtos/zephyr #31362: kconfiglib.py _save_old() may rename /dev/null -- replacing /dev/null with a file

domo141, 2021-01-15:
**Describe the bug**

the kconfiglib.py _save_old("/dev/null") may actually, unexpectly, succeed. 

afterwards /dev/null was opened for writing as a (new) file 

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. ls -l /dev/null
4. cat /dev/null

**Expected behavior**

/dev/null is still device node

**Impact**

/dev/null is a file, with kconfig content

**What impact does this issue have on your progress**

host code that opened /dev/null as config file got "garbage"

**Environment (please complete the following information):**

- a docker container (not as root, but probably enough capabilites to rename /dev/null)

zephyr v1.14 -- but after investigating main in 2021-01 it still can happen

** how to fix **

`if not os.path.isfile(path) return` to skip rename if path does not point to a file
(either directly, or via symlink)

nashif, 2021-01-19:
Can you please submit a PR with the proposed fix?

domo141, 2021-01-19:
Created (as usual, change was simple, comments and commit message hard)...

 #31438: based on commit msgs ulfalizer may (or may not) follow up soon...

nashif, 2021-01-23:
lowering priority as this happens only if /dev/null is writable or as root, which is not recommended.

PR #31438 (2021-01-28): kconfig: kconfiglib.py: Backup files only
The _save_old() to return early if <filename> is not file (or symlink
to a file).

This is simplest alternative to avoid attempt to rename /dev/null
(which could succeed).

This also keeps fifos (perhaps nonexistent potential usage but
this is nicer).

If <filename> were directory or socket, after shutil.copyfile(),
writing to the file (by caller, _write_config()), would fail.

Fixes #31362
commit 66d1b3ce (2021-01-28): kconfig: kconfiglib.py: Backup files only

The _save_old() to return early if <filename> is not file (or symlink
to a file).

This is simplest alternative to avoid attempt to rename /dev/null
(which could succeed).

This also keeps fifos (perhaps nonexistent potential usage but
this is nicer).

If <filename> were directory or socket, after shutil.copyfile(),
writing to the file (by caller, _write_config()), would fail.

Fixes #31362
closed 2021-01-28 by nashif