zephyrproject-rtos/zephyr #31376: Esp offload driver support server mode

lgl88911, 2021-01-16:
Now esp offload driver does not implement accept and listen socket API, It means the device that uses esp as wifi can only work in client state, but cannot become server.
Are there any plans to increase the implementation of accept and listen for esp offload in the near future? Thanks

mniestroj, 2021-01-19:
@lgl88911 I do not plan to implement accept/listen and don't know about anybody else doing it now. However patches are welcome!

lgl88911, 2021-01-20:
@mniestroj , Current client mode can do a lot of things for me. But in the future, I still hope to make esp offload work under the server, so I did some investigations, and think that esp at can be used to implement accept and listen，Can you give some thoughts？ Thanks！
AT+CIPSERVERMAXCONN ->listen
AT+CIPSERVER= 1 -> accept

There is no urgent need for server mode now. If in the future when I need server mode, esp offload is not yet supported, I will try to implement it.

tsvehagen, 2021-01-21:
So I did actually do server mode while implementing this driver but I decided to not include it. I can't remember the exact problem but it had something to do with synchronizing the link id's when using both accept and also doing connects.

I think I still have the original implementation somewhere, I'll have a look. I guess it could be commited as `EXPERIMENTAL` with some note that there might be some weird cases while using server and client simultaneously.

If no connections need to be initiated from device I think there should be no problems.

lgl88911, 2021-01-21:
@tsvehagen  I am glad to hear that you have implemented server mode, is it based on accpet and listen? If there are already accept and listen prototypes for esp offload, I would like to see the commit, whether as EXPERIMENTAL or other.
 I directly used the AT command to simply test on ESP32-WROVER_AT_Bin_V2.1.0.0 to make the server and client work at the same time, and it seems that there is no problem. Anyway, having runnable code is the first step, even if it has bugs, right?

maweicoltsmart, 2022-01-13:
> @tsvehagen I am glad to hear that you have implemented server mode, is it based on accpet and listen? If there are already accept and listen prototypes for esp offload, I would like to see the [...]

I also need to use esp-at as tcp server. Has there been progress on this issue?

tsvehagen, 2022-02-08:
Hi, sorry fot not answering earlier but I've been busy with other things :p anyway, the code was not in that good shape but I include the `ap_enable` and `ap_disable` below.

Not that this is quite old and won't be able to able as is. What's missing though is an implementation for `accept()`. IIRC when an incoming connection is established, a "<ID>,CONNECT" will be issued. The idea that I had was that when `accept` is called, we set the `accept_cb` and in the CONNECT command handler check if there is a ongoing accept by using the `accept_cb` and calling that one instead of the `connect_cb`. But I believe there actually was no way of knowing if it was from an incoming or outgoing connection and that's when you get into trouble if using both.

```
static int esp8266_mgmt_ap_enable(struct device *dev,
				  struct wifi_connect_req_params *params)
{
	char cmd[sizeof("AT+CWSAP_CUR=\"\",\"\",xx,x") + WIFI_SSID_MAX_LEN +
		 WIFI_PSK_MAX_LEN];
	struct esp8266_data *data = dev->driver_data;
	int ecn = 0, len, ret;

	ret = modem_cmd_send(&data->mctx.iface, &data->mctx.cmd_handler,
			     NULL, 0, "AT+CWMODE_CUR=3", &data->sem_response,
			     MDM_CMD_TIMEOUT);
	if (ret < 0) {
		LOG_ERR("Failed to enable AP mode, ret %d", ret);
		return ret;
	}

	len = snprintk(cmd, sizeof(cmd), "AT+CWSAP_CUR=\"");
	memcpy(&cmd[len], params->ssid, params->ssid_length);
	len += params->ssid_length;

	if (params->security == WIFI_SECURITY_TYPE_PSK) {
		len += snprintk(&cmd[len], sizeof(cmd) - len, "\",\"");
		memcpy(&cmd[len], params->psk, params->psk_length);
		len += params->psk_length;
		ecn = 3;
	} else {
		len += snprintk(&cmd[len], sizeof(cmd) - len, "\",\"");
	}

	snprintk(&cmd[len], sizeof(cmd) - len, "\",%d,%d", params->channel,
		 ecn);

	ret = modem_cmd_send(&data->mctx.iface, &data->mctx.cmd_handler,
			     NULL, 0, cmd, &data->sem_response,
			     MDM_CMD_TIMEOUT);

	return ret;
}

static int esp8266_mgmt_ap_disable(struct device *dev)
{
	struct esp8266_data *data = dev->driver_data;
	int ret;

	ret = modem_cmd_send(&data->mctx.iface, &data->mctx.cmd_handler,
			     NULL, 0, "AT+CWMODE_CUR=1", &data->sem_response,
			     MDM_CMD_TIMEOUT);

	return ret;
}

static const struct net_wifi_mgmt_offload esp8266_api = {
	.iface_api.init = esp8266_iface_init,
	.scan		= esp8266_mgmt_scan,
	.connect	= esp8266_mgmt_connect,
	.disconnect	= esp8266_mgmt_disconnect,
	.ap_enable	= esp8266_mgmt_ap_enable,
	.ap_disable	= esp8266_mgmt_ap_disable,
};
```

maweicoltsmart, 2022-02-12:
@[tsvehagen](https://github.com/tsvehagen)
Thank you for replying to me while being so busy. Do you mean the problem occurs when TCP Client and TCP Server are used at the same time? In fact, TCP Client and TCP Server may not be used at the same time in a certain project. So I think, you can do a special process in the program to avoid using TCP Client and TCP Server at the same time, for example: when the first TCP Client socket connection is created, the calls to listen and accept both return errors and give log prompts ; When listen is called, the connect call to the TCP Client returns an error and gives a log prompt. In this case, as long as we don't use TCP Client and TCP Server at the same time, there will be no problem, instead of applications that have no way to use TCP Server as they are now.

carlescufi, 2022-03-08:
@sylvioalves is this something you can take a look at?

kartben, 2026-10-30:
Closing as a duplicate of #51642, which tracks bind/accept support for server mode in the ESP-AT offload driver. Please follow up there.