zephyrproject-rtos/zephyr #31400: Extending `zephyr,code-partition` with `zephyr,code-header-size`

Laczen, 2021-01-18:
**Is your enhancement proposal related to a problem? Please describe.**

At the moment it is possible to generate images for a specific partition by using `zephyr,code-partition = &partition` in the dts. 

However it is not possible to specify the size of the header. This is specified by e.g. selecting `CONFIG_USE_MCUBOOT = y`.

If it would be possible to specify the header size in the dts by using (e.g. `zephyr,code-header-size = <0x200>` the generation of images for any bootloader would be simplified by creating an appropriate overlay.

The zephyr default that does not specify `zephyr,code-partition` or `zephyr,code-header-size` would generate an image for the complete `flash0`.

A dts that specifies `zephyr,code-partition` but not `zephyr,code-header-size` would generate a image for `code-partition` starting at the start of the partition. Using a dts that specifies an image partition and a storage partition would avoid that the image overflows in the storage partition. This could be the default dts for boards that are using the storage area.

A dts that specifies `zephyr, code-partition` and `zephyr,code-header-size` would generate a image for `code-partition` but starting after the `code-header-size`. This could be specified in a `mcuboot` dts that includes (redefines) partitioning and redefines `zephyr,code-partition` and sets `zephyr,code-header-size = <0x200>`.

**Describe alternatives you've considered**
Alternative solutions have been proposed in #27974 and #31073 (side benefit).

de-nordic, 2021-01-18:
What about CONFIG_ROM_START_OFFSET ? If the CONFIG_USE_MCUBOOT is not set this value is 0 and is configurable.

Laczen, 2021-01-18:
@de-nordic, thanks for the feedback.

The idea is to remove everything from Kconfig, only use dts (and overlays). CONFIG_ROM_START_OFFSET would be calculated from `zephyr,code-header-size` (if set).

de-nordic, 2021-01-18:
I'm not sure, but this https://github.com/zephyrproject-rtos/zephyr/pull/29794#pullrequestreview-568461266 may apply here.

>The idea is to remove everything from Kconfig, only use dts (and overlays).

I am in favour of the idea, but for everything that can be considered hardware parameters.
The problem with `zephyr,code-header-size` is that this is actually software parameter rather than something that applies to hardware configuration. I am not sure about this.

Laczen, 2021-01-18:
@de-nordic, I think this could be considered a hardware parameter, it is something that the "hardware" (bootloader) expects, it is not related to the image (software) itself. But in this case the line is difficult to draw.

@galak, what is your opinion on the proposal ? It is true that it could be considered a software parameter, but it might be the simplest way to support whatever bootloader and to resolve #27471.

Laczen, 2021-01-18:
@ioannisg, fyi

Laczen, 2021-01-19:
@mbolivar-nordic, what is your opinion on the proposal ?

Laczen, 2021-01-21:
I am closing this as there is an alternative proposal.

closed 2021-01-21 by Laczen