zephyrproject-rtos/zephyr #31408: Improve subsys/settings documentation and sample code

sw, 2021-01-18:
I was looking into using the settings subsystem and was quite confused by the documentation and sample code.

The following verbs are used in the code and documentation when referring to operations on setting values:
* apply
* commit
* dump
* export
* get
* load
* read
* save
* set
* write

I suspect that some of these are synonyms and some - importantly - aren't, but it isn't really clear.

The distinction between "source" and "destination" also depends on whether we are loading data from non-volatile storage or writing into it. I'm assuming that "target" and "destination" refer to the same thing at least?

What is the "runtime" backend and why would I want to use it or not use it?

From the "Garbage collection" section in the docs:
https://github.com/zephyrproject-rtos/zephyr/blob/b1080ea036caf0ccd1141082030d700405ab8e3c/doc/reference/settings/index.rst#L138-L140
This sounds as if the settings subsystem will randomly declare some of my data "garbage" and delete them. That doesn't sound like something anybody would want, or am I misunderstanding this?

Commenting of the sample code is weird in places, e.g.:
https://github.com/zephyrproject-rtos/zephyr/blob/b1080ea036caf0ccd1141082030d700405ab8e3c/samples/subsys/settings/src/main.c#L24

sw, 2021-01-18:
As an example of what I mean by the verbs:

To me, these are synonyms respectively:
* get = load
* set = store

But the docs on `h_set` says:
https://github.com/zephyrproject-rtos/zephyr/blob/b1080ea036caf0ccd1141082030d700405ab8e3c/doc/reference/settings/index.rst#L40-L43

So here set = load?

sw, 2021-01-18:
Could `export_func` be defined as a `typedef`, same as for `settings_read_cb`? That would simplify the definition of `settings_handler` and `settings_handler_static`.

sw, 2021-01-19:
Here's how I understand the settings subsystem after playing around with it for a bit. A diagram like this may be useful in the documentation. Of course mine may still be wrong, and does not include the runtime backend nor multiple backends. 

Loading:
```
application          settings                         settings
     |              subsystem                         backend
     |                  |                                |
     | settings_load()  |                                |
     |----------------->|          csi_load()            |
     |                  |------------------------------->|
     |                  |                                |
     |                  | settings_call_set_handler(...) |
     |    h_set(...)    |<-------------------------------|
     |<-----------------|                                |
     |                  | settings_call_set_handler(...) |
     |    h_set(...)    |<-------------------------------|
     |<-----------------|                                |
     |                  |                                |
     |    h_commit()    |                                |
     |<-----------------|                                |
```

Saving:
```
application          settings                         settings
     |              subsystem                         backend
     |                  |                                |
     | settings_save()  |                                |
     |----------------->|      csi_save_start(...)       |
     |                  |------------------------------->|
     |  h_export(...)   |                                |
     |<-----------------|                                |
     |                  |                                |
     | export_func(...) |                                |
     |----------------->|         csi_save(...)          |
     |                  |------------------------------->|
     | export_func(...) |                                |
     |----------------->|         csi_save(...)          |
     |                  |------------------------------->|
     |                  |                                |
     |                  |        csi_save_end(...)       |
     |                  |------------------------------->|
```

Laczen, 2021-01-19:
> I was looking into using the settings subsystem and was quite confused by the documentation and sample code.
> I suspect that some of these are synonyms and some - importantly - aren't, but it isn't really clear.

@sw, the settings subsystem is a quite complex system. The documentation might indeed be lacking to easily understand it, the naming of the subroutines might be strange sometimes but let's take the example of `h_set`: what is does is provide a routine to set a variable and this routine is called when `settings_load()` is called. The effect is the loading of the variable, but naming it `load` would not be correct, it doesn't load anything.

> 
> ~The distinction between "source" and "destination" also depends on whether we are loading data from non-volatile storage or writing into it. I'm assuming that "target" and "destination" refer to the same thing at least?~ After re-reading those sections it seems clearer to me now: these word only refer to the non-volatile storage, not the copy in RAM.
> 
> What is the "runtime" backend and why would I want to use it or not use it?
> 

The "runtime" backend can be used make your own custom backend out of the zephyr tree.

> From the "Garbage collection" section in the docs:
> https://github.com/zephyrproject-rtos/zephyr/blob/b1080ea036caf0ccd1141082030d700405ab8e3c/doc/reference/settings/index.rst#L138-L140
>
>
> This sounds [...]

You are misunderstanding this. Whenever an update of a settings is written to a backend a new entry is written, the garbage collection will remove the old entries as well as any deleted entries.

Your schematic is basically indeed what is done, but it is missing some key features like filtered loading, multiple backends, ...

nvlsianpu, 2021-01-19:
@sw
> Could export_func be defined as a typedef, same as for settings_read_cb? That would simplify the definition of settings_handler and settings_handler_static.

I think yes, you can open a PR for that. We are using kernel like CS here, which mean zephyr-project recomend not to define  types for need of a while, but in this case I think might be usefully.

Nice diagrams BTW.

Laczen, 2021-01-19:
@sw, it is also worth noting that the set, get, commit, export functionality are all optional and can be set to null functions.

In the simplest case just the set function would be defined, allowing loading of variables while saving is done using `settings_save_one()`. A addition to that would be to define the commit function allowing the "use" of variables after all variables have been loaded.

Also it is possible to only use direct loading: i.e. you would never call settings_load(), but instead you call `settings_load_subtree()` with the name of the variable (if the set function is defined) to load or `settings_load_subtree_direct()` where you provide your own set routine to `settings_load_subtree_direct()` and you don't have to register any static or dynamic handler.

mikedudgeon, 2022-01-17:
As a new Zephyr developer but very experienced embedded guy, I agree that the settings documentation is poor even as of 2.7.  I have found the rest of the docs very good, but not the settings.

psztoch, 2024-08-21:
Additionally, the example file `samples/subsys/settings/src/main.c` has a few bugs in my opinion. Something is commented and something else is displayed/done.