zephyrproject-rtos/zephyr #31415: [RFC] ARC: interrupt management rework

evgeniy-paltsev, 2021-01-18:
## Problem description

In case of SMP system interrupt infrastructure implementation for ARC has issues.

We have per-core interrupt controllers private for each core. We setup interrupt controller only on CPU0 and route all external interrupts to CPU0. 
That makes our SMP system not so symmetric, but things are getting worse when we start using dynamic interrupts - as when task which is enabling interrupt can be executed on CPU != 0, so we enable interrupt on, lets say, CPU1, however IDU will route interrupt on CPU0 for processing (where it is still disabled). See #31298 for example.
Quite similar situation with dynamic interrupt disabling is also possible.

As I understand the per-core private interrupts (like timer0/1 interrupts) are working now by lucky chance - as we have all interrupts enabled in per-core interrupt controller and we don't setup it for CPU != 0.

## How to fix - option 1:
We allow interrupt handling only on CPU0 (as we have it now)
We allow interrupt management function to be executed only on CPU0.

Pros:
- Easy to implement.

Cons:
- 'not so symmetric' SMP
- noticeably inconvenient for users if they use dynamic interrupts

Require clarifications:
- Is the chance that such 'not so symmetric' SMP implementation may cause performance degradation?

## How to fix - option 2:
We allow interrupt handling on all CPUs, so we do initial interrupt controller setup on all CPUs and enable round-robin interrupt distribution mode in IDU.
The dynamic interrupt enabling/disabling must be done on all CPUs which require invocations some code with ICI (inter-core-interrupts)

Pros:
- real SMP
- all interrupts can be dynamic, even per-core. But do we really need this?

Cons:
- simple function like enable_interrupt becomes quite heavy and requires inter-core-interrupts invocation.
- with ICI invocation it is really easy create new issues (which are badly reproducible and badly debuggable) in other corner cases.

Require clarifications:
- What would we break by adding ICI for interrupt management on per-cpu interrupt controllers.

## How to fix - option 3:
We allow interrupt handling on all CPUs, so we do initial interrupt controller setup on all CPUs and enable round-robin interrupt distribution mode in IDU.
We manage interrupts (enable/disable) with IDU instead of per-core interrupt controller.
The common interrupts can be software generated with CMD_IDU_GEN_CIRQ, so test for dynamic interrupts still can be implemented.

Pros:
- real SMP
- IDU is global, so we don't need inter-core-interrupts for interrupts management
- approach proven by Linux kernel

Cons:
- only common interrupts can be dynamic (do we really need dynamic per-core interrupts?)
- different interrupt controller used for UP and SMP

Require clarifications:
- what should we do with interrupt priority - it is still controlled by per-core interrupt controller.The Linux kernel uses single interrupt priority.

evgeniy-paltsev, 2021-01-18:
@abrodkin @ruuddw @IRISZZW Let's discuss this.

evgeniy-paltsev, 2021-01-23:
I've investigated a bit about `arch_irq_{disable|enable}` usage. In `include/irq.h` we do define

```
#define irq_enable(irq) arch_irq_enable(irq)
#define irq_disable(irq) arch_irq_disable(irq)
```

And rest of the Zephyr code actively uses `irq_{disable|enable}`. So we can't simply restrict interrupt management function to be executed only on CPU0 as it will break drivers/subsystems.

As drivers/subsystems control external interrupts only the option 3 should be OK for them. However there is at least one place (`tests/kernel/context`) where we enable and disable ARC tick timer interrupt (IRQ_TIMER0). This corner case needs to be investigated as we can't control timer interrupt with IDU.

evgeniy-paltsev, 2021-01-24:
Thoughts about issue with interrupt priority in case of 'option 3':

We can't control interrupt priority with IDU so we have to configure interrupt priority via per-core interrupt controller.
Generally it should be OK as we need to do that only once - at the per-core interrupt controller init step.
There are several places where you can shoot in the leg, so the initialization needs to be done carefully. For example we need to enable round-robin interrupt distribution mode in IDU only after the interrupt controllers of each core are initialized.

The things are getting more complicated in case of dynamic interrupts as we can't configure interrupt priority at init step of the per-core interrupt controller. This means that we do configuration on each core in the moment of `irq_connect_dynamic` call. That obviously require ICI usage.

As of today dynamic interrupt subsystem has limitation:
> Removing or re-configuring a dynamic interrupt is currently unsupported.

so, most likely, the ICI usage shouldn't cause the problems
 * we do it once per interrupt - it shouldn't bring performance penalty
 * we do it on the interrupt disabled at the moment of configuration - there shouldn't be tricky synchronization issues

ruuddw, 2021-01-25:
> So we can't simply restrict interrupt management function to be executed only on CPU0 as it will break drivers/subsystems. **IOW - option 1 is no go**.

Why not? it is 'just' that interrupt functions running on other cores would need to forward the request to CPU 0 with an IPI request (for external interrupts). Overhead of this would probably not be an issue, as irq_enable/disable() is typically used at (de)initialization only. But, it is more complex than you initially might have thought.

evgeniy-paltsev, 2021-10-04:
WIP branch: https://github.com/evgeniy-paltsev/zephyr/tree/rff-interrupt-management-rework-v3