zephyrproject-rtos/zephyr #31435: Dummy device drivers for DT test bindings

mbolivar-nordic, 2021-01-19:
**Is your enhancement proposal related to a problem? Please describe.**

We are moving to a world where devices are increasingly resolved at link time from devicetree node identifiers, not at runtime from name strings using device_get_binding().

In particular, we now have macros which return pointers to global device pointers that resolve at link time: DEVICE_DT_GET() (added in e643ee26a71) and DEVICE_DT_INST_GET() (added in 3a83f0ec69e). These return pointers to devices created using DEVICE_DT_DEFINE() and DEVICE_DT_INST_DEFINE(). We have also deprecated DEVICE_AND_API_INIT().

This makes it possible to improve existing device driver implementations that depend on other drivers. We can write them with less overhead and boilerplate using DEVICE_DT_[INST_]GET() instead of saving devicetree label properties as strings and calling device_get_binding() at runtime.

However, we have a problem build-testing those improvements in tests/drivers/build_all. The problem is that any drivers using DEVICE_DT_GET() or the INST variant to get devices they depend on (like a SPI bus device used by a sensor driver) will fail to build. The failure is caused by the bus nodes in build_all's devicetree overlay referring to nodes with dummy compatibles like `vnd,spi`, `vnd,i2c`, etc. defined in dts/bindings/test. These compatibles currently lack device drivers, so the bus nodes don't have any corresponding device structures. Attempting to use DEVICE_DT_GET() with such nodes thus causes link errors.

**Describe the solution you'd like**

Create dummy device drivers which instantiate struct devices for any `vnd,foo` compatibles needed to build-test device drivers.

Pull request https://github.com/zephyrproject-rtos/zephyr/pull/30536 does this for SPI and I2C. This issue tracks that, as well as finishing the job for for gpio and serial. Other `vnd` compatibles which need dummy drivers should be evaluated on a case by case basis.

**Describe alternatives you've considered**

No other solutions seem practical. The only alternative is to refer to "real" hardware in the SoC DTS or board DTS, which is a maintenance nightmare for a simple build test.

pch-jp, 2021-02-03:
Can I vote this up?  I've been through this same situation.

henrikbrixandersen, 2022-11-02:
I would suggest we consider using FFF-based fake drivers for this kind of testing (see f30a5969d0302745b9afeb3c3c7946b434dbdbb5 and 10c37a3ab4e0cc842ae73cb2c4c6128893582062 for an example).

mbolivar-nordic, 2022-11-05:
@henrikbrixandersen assuming we're still setting up the devices in DT with some sort of fake or vnd compatible, no objections.

dcpleung, 2024-02-08:
UART already has a test driver.

abenezerarg, 2025-06-04:
+1 for clock and pinctrl as well, this suggestion is extremely beneficial for unit testing drivers that invoke HW specific API