zephyrproject-rtos/zephyr #31436: compliance script broken

pabigot, 2021-01-19:
After 251648afb458b5593bbd8216e6fc169b0a870e92 I'm getting the following on the master branch:

```
tirzah[31]$ scripts/ci/check_compliance.py -c HEAD^..HEAD
Running checkpatch       tests in /scratch/zephyr/zp1/zephyr ...
Running Kconfig          tests in /scratch/zephyr/zp1/zephyr ...
Python exception in `scripts/ci/check_compliance.py`:

```
Traceback (most recent call last):
  File "/scratch/zephyr/zp1/zephyr/scripts/kconfig/kconfiglib.py", line 2189, in _enter_file
    self._readline = self._open(filename, "r").readline
  File "/scratch/zephyr/zp1/zephyr/scripts/kconfig/kconfiglib.py", line 3922, in _open
    open(filename, mode, encoding=self._encoding)
IsADirectoryError: [Errno 21] Is a directory: '/scratch/zephyr/zp1/zephyr/'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "scripts/ci/check_compliance.py", line 1081, in main
    n_fails = _main(args)
  File "scripts/ci/check_compliance.py", line 1032, in _main
    test.run()
  File "scripts/ci/check_compliance.py", line 229, in run
    kconf = self.parse_kconfig()
  File "scripts/ci/check_compliance.py", line 335, in parse_kconfig
    return kconfiglib.Kconfig()
  File "/scratch/zephyr/zp1/zephyr/scripts/kconfig/kconfiglib.py", line 947, in __init__
    self._init(filename, warn, warn_to_stderr, encoding)
  File "/scratch/zephyr/zp1/zephyr/scripts/kconfig/kconfiglib.py", line 1085, in _init
    self._parse_block(None, self.top_node, self.top_node).next = None
  File "/scratch/zephyr/zp1/zephyr/scripts/kconfig/kconfiglib.py", line 2989, in _parse_block
    prev = self._parse_block(None, parent, prev)
  File "/scratch/zephyr/zp1/zephyr/scripts/kconfig/kconfiglib.py", line 3028, in _parse_block
    self._parse_block(_T_ENDMENU, node, node)
  File "/scratch/zephyr/zp1/zephyr/scripts/kconfig/kconfiglib.py", line 2989, in _parse_block
    prev = self._parse_block(None, parent, prev)
  File "/scratch/zephyr/zp1/zephyr/scripts/kconfig/kconfiglib.py", line 2989, in _parse_block
    prev = self._parse_block(None, parent, prev)
  File "/scratch/zephyr/zp1/zephyr/scripts/kconfig/kconfiglib.py", line 3028, in _parse_block
    self._parse_block(_T_ENDMENU, node, node)
  File "/scratch/zephyr/zp1/zephyr/scripts/kconfig/kconfiglib.py", line 2988, in _parse_block
    self._enter_file(filename)
  File "/scratch/zephyr/zp1/zephyr/scripts/kconfig/kconfiglib.py", line 2192, in _enter_file
    raise _KconfigIOError(
kconfiglib._KconfigIOError: /tmp/Kconfig.modules:2: Could not open '/scratch/zephyr/zp1/zephyr/' (in 'osource "$(ZEPHYR_HAL_NORDIC_KCONFIG)"') (EISDIR: Is a directory)

```
Traceback (most recent call last):
  File "/scratch/zephyr/zp1/zephyr/scripts/kconfig/kconfiglib.py", line 2189, in _enter_file
    self._readline = self._open(filename, "r").readline
  File "/scratch/zephyr/zp1/zephyr/scripts/kconfig/kconfiglib.py", line 3922, in _open
    open(filename, mode, encoding=self._encoding)
IsADirectoryError: [Errno 21] Is a directory: '/scratch/zephyr/zp1/zephyr/'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "scripts/ci/check_compliance.py", line 1108, in <module>
    main()
  File "scripts/ci/check_compliance.py", line 1081, in main
    n_fails = _main(args)
  File "scripts/ci/check_compliance.py", line 1032, in _main
    test.run()
  File "scripts/ci/check_compliance.py", line 229, in run
    kconf = self.parse_kconfig()
  File "scripts/ci/check_compliance.py", line 335, in parse_kconfig
    return kconfiglib.Kconfig()
  File "/scratch/zephyr/zp1/zephyr/scripts/kconfig/kconfiglib.py", line 947, in __init__
    self._init(filename, warn, warn_to_stderr, encoding)
  File "/scratch/zephyr/zp1/zephyr/scripts/kconfig/kconfiglib.py", line 1085, in _init
    self._parse_block(None, self.top_node, self.top_node).next = None
  File "/scratch/zephyr/zp1/zephyr/scripts/kconfig/kconfiglib.py", line 2989, in _parse_block
    prev = self._parse_block(None, parent, prev)
  File "/scratch/zephyr/zp1/zephyr/scripts/kconfig/kconfiglib.py", line 3028, in _parse_block
    self._parse_block(_T_ENDMENU, node, node)
  File "/scratch/zephyr/zp1/zephyr/scripts/kconfig/kconfiglib.py", line 2989, in _parse_block
    prev = self._parse_block(None, parent, prev)
  File "/scratch/zephyr/zp1/zephyr/scripts/kconfig/kconfiglib.py", line 2989, in _parse_block
    prev = self._parse_block(None, parent, prev)
  File "/scratch/zephyr/zp1/zephyr/scripts/kconfig/kconfiglib.py", line 3028, in _parse_block
    self._parse_block(_T_ENDMENU, node, node)
  File "/scratch/zephyr/zp1/zephyr/scripts/kconfig/kconfiglib.py", line 2988, in _parse_block
    self._enter_file(filename)
  File "/scratch/zephyr/zp1/zephyr/scripts/kconfig/kconfiglib.py", line 2192, in _enter_file
    raise _KconfigIOError(
kconfiglib._KconfigIOError: /tmp/Kconfig.modules:2: Could not open '/scratch/zephyr/zp1/zephyr/' (in 'osource "$(ZEPHYR_HAL_NORDIC_KCONFIG)"') (EISDIR: Is a directory)
```

pabigot, 2021-01-19:
It appears that 6d3cdf5b4aa33689f8cfacb7a7a4cb3d9269b91a inadvertently fixed this.

pabigot, 2021-01-19:
No, I'm still seeing it.

mbolivar-nordic, 2021-01-19:
Doesn't repro for me.

```
$ ./scripts/ci/check_compliance.py -c HEAD^..HEAD
Running checkpatch       tests in /home/mbolivar/zp/zephyr ...
Running Kconfig          tests in None ...
Running Codeowners       tests in /home/mbolivar/zp/zephyr ...
Running Nits             tests in /home/mbolivar/zp/zephyr ...
Running Gitlint          tests in /home/mbolivar/zp/zephyr ...
Running pylint           tests in /home/mbolivar/zp/zephyr ...
Running Identity         tests in /home/mbolivar/zp/zephyr ...
WARNING : Skipped Kconfig, Not a Zephyr tree (ZEPHYR_BASE unset)

Complete results in compliance.xml
$ git log -1
commit 13aa4b2f9ea0c474c8c809e69a2b806b3913f5c1 (HEAD -> master, origin/master)
Author: Hubert Miś <hubert.mis@nordicsemi.no>
Date:   Sun Jan 17 23:09:28 2021 +0100

    net: coap: define Content-Format option values
    
    CoAP protocol defines registry of Content-Format option values.
    This patch adds this enumeration to coap header file to make it
    available to all applications using CoAP protocol. It modifies
    code using CoAP service to use new enumeration.
    
    Signed-off-by: Hubert Miś <hubert.mis@nordicsemi.no>
```

henrikbrixandersen, 2021-01-19:
I can reproduce the issue here.

```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.5 LTS
Release:	18.04
Codename:	bionic
```

pabigot, 2021-01-19:
A clean revert would probably require reverting:
* #31115 then
* #31224
which involves seven commits.

pabigot, 2021-01-19:
In case it's OS dependent:
```
tirzah[2195]$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.1 LTS
Release:	20.04
Codename:	focal
```

pabigot, 2021-01-19:
Per comment on slack from @mbolivar-nordic this may be related to having `ZEPHYR_BASE` set.

tejlmand, 2021-01-20:
Looking into a fix.

Somehow CI didn't fail the pr, even though compliance failed.
https://github.com/zephyrproject-rtos/zephyr/runs/1727523984#step:7:157

PR #31446 (2021-01-20): scripts: check_compliance.py: include known Zephyr Kconfig glue code
Fixes: #31436

This fix adds ZEPHYR_<MODULE_NAME>_KCONFIG variable to compliance check.
This ensures that Zephyr Kconfig glue code is included in the
compliance check, and thus remove the issue reported in #31436.
commit 3361b154 (2021-01-20): scripts: check_compliance.py: include known Zephyr Kconfig glue code

Fixes: #31436

This fix adds ZEPHYR_<MODULE_NAME>_KCONFIG variable to compliance check.
This ensures that Zephyr Kconfig glue code is included in the
compliance check, and thus remove the issue reported in #31436.
closed 2021-01-20 by carlescufi