zephyrproject-rtos/zephyr #31439: nrf5340dk_nrf5340_cpunet configuring incomplete

mengxianglinx, 2021-01-20:
**Describe the bug**
Run test case with platform nrf5340dk_nrf5340_cpunet, build error occur.

**To Reproduce**
Steps to reproduce the behavior:
1. run "twister -T tests/subsys/power/power_mgmt -p nrf5340dk_nrf5340_cpunet
2. compile failure because of "configuring incomplete", see buld.log

**Expected behavior**
Test cases PASS

**Logs and console output**
In build.log:
warning: HAS_NORDIC_DRIVERS (defined at modules/hal_nordic/Kconfig:7) has direct dependencies 0 with value n, but is currently being y-selected by the following symbols:
   - SOC_SERIES_NRF53X (defined at soc/arm/nordic_nrf/nrf53/Kconfig.series:6), with value y, direct dependencies <choice> (value: y), and select condition <choice> (value: y)
 warning: HAS_NRFX (defined at modules/hal_nordic/nrfx/Kconfig:4) has direct dependencies 0 with value n, but is currently being y-selected by the following symbols:
  - SOC_SERIES_NRF53X (defined at soc/arm/nordic_nrf/nrf53/Kconfig.series:6), with value y, direct dependencies <choice> (value: y), and select condition <choice> (value: y)
warning: NRFX_CLOCK (defined at modules/hal_nordic/nrfx/Kconfig:14) has direct dependencies HAS_HW_NRF_CLOCK && HAS_NRFX && 0 with value n, but is currently being y-selected by the following symbols:
  - CLOCK_CONTROL_NRF (defined at drivers/clock_control/Kconfig.nrf:13), with value y, direct dependencies SOC_COMPATIBLE_NRF && !CLOCK_CONTROL_NRF_FORCE_ALT && CLOCK_CONTROL (value: y), and select condition     SOC_COMPATIBLE_NRF && !CLOCK_CONTROL_NRF_FORCE_ALT && CLOCK_CONTROL (value: y)
 warning: NRFX_CLOCK_LFXO_TWO_STAGE_ENABLED (defined at modules/hal_nordic/nrfx/Kconfig:18) has direct dependencies NRFX_CLOCK && HAS_NRFX && 0 with value n, but is currently being y-selected by the following     symbols:
 - CLOCK_CONTROL_NRF_K32SRC_XTAL (defined at drivers/clock_control/Kconfig.nrf:37), with value y, direct dependencies <choice CLOCK_CONTROL_NRF_SOURCE> (value: y), and select condition !SOC_SERIES_BSIM_NRFXX     && <choice CLOCK_CONTROL_NRF_SOURCE> (value: y)
 warning: NRFX_DPPI (defined at modules/hal_nordic/nrfx/Kconfig:26, drivers/serial/Kconfig.nrfx:350) has direct dependencies (HAS_HW_NRF_DPPIC && HAS_NRFX && 0) || (HAS_HW_NRF_DPPIC && (UART_0_NRF_HW_ASYNC ||     UART_1_NRF_HW_ASYNC || UART_2_NRF_HW_ASYNC || UART_3_NRF_HW_ASYNC) && UART_NRFX && SERIAL) with value n, but is currently being y-selected by the following symbols:
  - UART_0_ENHANCED_POLL_OUT (defined at drivers/serial/Kconfig.nrfx:39), with value y, direct dependencies UART_0_NRF_UARTE && (UART_0_NRF_UART || UART_0_NRF_UARTE) && UART_NRFX && SERIAL (value: y), and sel    ect condition HAS_HW_NRF_DPPIC && UART_0_NRF_UARTE && (UART_0_NRF_UART || UART_0_NRF_UARTE) && UART_NRFX && SERIAL (value: y)
error: Aborting due to Kconfig warnings

**Additional context**
Add any other context about the problem here.

anangl, 2021-01-21:
Try running `west update` first.

mengxianglinx, 2021-01-21:
After "west update", run "twister -T tests/subsys/power/power_mgmt -p nrf5340dk_nrf5340_cpunet":
 In file included from /media/dskpool/ws-xianglin/zephyrproject/zephyr/soc/arm/nordic_nrf/nrf53/power.c:11:
/media/dskpool/ws-xianglin/zephyrproject/modules/hal/nordic/nrfx/hal/nrf_regulators.h:138:50: error: unknown type name 'NRF_REGULATORS_Type'; did you mean 'NRF_REGULATORS_H__'?

closed 2021-01-21 by mengxianglinx