zephyrproject-rtos/zephyr #31447: MQTT idling gets disconnected when using TCP2

arapostol, 2021-01-20:
**Describe the bug**
MQTT disconnects with -57 (ENOTCONN) at random times when idling, even within seconds after receiving a PINGRESP. After that reconnection may fail. Socket logs show a peer-closed socket.
The problem disappears when using TCP1. Tested with and without TLS.

**To Reproduce**
Steps to reproduce the behavior:
1. modify mqtt_publisher sample to idle only, by setting
```
while (connected) {
		rc = process_mqtt_and_sleep(&client_ctx, APP_SLEEP_MSECS);
		SUCCESS_OR_BREAK(rc);
}
```
2. connect to a broker
3. wait a few minutes to a couple of hours

**Expected behavior**
To remain connected or disconnect on network / broker error

**Impact**
Huge for data consumption if using with TLS on a GSM.

**Logs and console output**
> [09:36:26] I: tls_init: 0 <OK>
> [09:36:26] I: attempting to connect:
> [09:36:27] I: MQTT client connected!
> [09:36:27] I: try_to_connect: 0 <OK>
> [09:37:28] I: PINGRESP packet
> [09:38:29] I: PINGRESP packet
> ---PINGRESP LOGS OMITTED---
> [11:02:14] I: PINGRESP packet
> [11:03:15] I: PINGRESP packet
> [11:04:16] I: PINGRESP packet
> [11:04:17] I: MQTT client disconnected -57
> [11:04:17] I: mqtt_input: -57 <ERROR>
> [11:04:17] I: mqtt_disconnect: -57 <ERROR>
> [11:04:17] I: Bye!
> [11:04:22] I: attempting to connect:
> [11:04:23] I: MQTT client connected!
> [11:04:23] I: try_to_connect: 0 <OK>
> [11:05:24] I: PINGRESP packet
> [11:06:25] I: PINGRESP packet
> [11:07:25] I: PINGRESP packet
> ---PINGRESP LOGS OMITTED---
> [12:50:25] I: PINGRESP packet
> [12:51:25] I: PINGRESP packet
> [12:51:27] I: MQTT client disconnected -57
> [12:51:27] I: mqtt_input: -57 <ERROR>
> [12:51:27] I: mqtt_disconnect: -57 <ERROR>
> [12:51:27] I: Bye!
> [12:51:32] I: attempting to connect:
> [12:51:33] I: MQTT client connected!
> [12:51:33] I: try_to_connect: 0 <OK>
> [12:52:34] I: PINGRESP packet
> [12:53:34] I: PINGRESP packet
> ---PINGRESP LOGS OMITTED---
> [13:23:51] I: PINGRESP packet
> [13:24:52] I: PINGRESP packet
> [13:24:54] I: MQTT client disconnected -57
> [13:24:54] I: mqtt_input: -57 <ERROR>
> [13:24:54] I: mqtt_disconnect: -57 <ERROR>
> [13:24:54] I: Bye!
> [13:24:59] I: attempting to connect:
> [13:25:00] I: MQTT client disconnected -53
> [13:25:00] I: MQTT client disconnected -53
> [13:25:01] I: MQTT client disconnected -53
> [13:25:02] I: MQTT client disconnected -53
> [13:25:03] I: MQTT client disconnected -53
> [13:25:03] I: MQTT client disconnected -53
> [13:25:04] I: MQTT client disconnected -53
> [13:25:05] I: MQTT client disconnected -53
> [13:25:06] I: MQTT client disconnected -53
> [13:25:06] I: MQTT client disconnected -53
> [13:25:06] I: try_to_connect: -22 <ERROR>
> [13:25:11] I: attempting to connect:
> [13:25:12] I: MQTT client disconnected -53
> [13:25:13] I: MQTT client disconnected -53
> [13:25:14] I: MQTT client disconnected -53
> [13:25:14] I: MQTT client disconnected -53
> [13:25:15] I: MQTT client disconnected -53
> [13:25:16] I: MQTT client disconnected -53
> [13:25:16] I: MQTT client disconnected -53
> [13:25:17] I: MQTT client disconnected -53
> [13:25:18] I: MQTT client disconnected -53
> [13:25:19] I: MQTT client connected!
> [13:25:19] I: try_to_connect: 0 <OK>
> [13:26:19] I: PINGRESP packet
> [13:27:20] I: PINGRESP packet
> ---PINGRESP LOGS OMITTED---
> [13:35:25] I: PINGRESP packet
> [13:36:25] I: PINGRESP packet
> [13:36:27] I: MQTT client disconnected -57
> [13:36:27] I: mqtt_input: -57 <ERROR>
> [13:36:27] I: mqtt_disconnect: -57 <ERROR>
> [13:36:27] I: Bye!
> [13:36:32] I: attempting to connect:
> [13:36:33] I: MQTT client connected!
> [13:36:33] I: try_to_connect: 0 <OK>
> [13:37:34] I: PINGRESP packet

**Environment (please complete the following information):**
 - OS: Linux, Windows
 - Toolchain: Zephyr SDK-0.11.3
 - Commit SHA: 02819489a22857bd2b8fbc5460a39a02de9215b1

**Additional context**
The problem also exists at zephyr-v2.4.0. Also tested with platformio.

jukkar, 2021-01-20:
@rlubos have you seen these kind of issues in your tests?

rlubos, 2021-01-20:
No, I haven't hit this issue before. I can try to reproduce and monitor the traffic with wireshark to see what's going on.

rlubos, 2021-01-21:
I had the sample running for over a dozen of hours in total, idling, and I was not able to reproduce the issue (I was testing with a local `mosquitto` instance and `qemu_x86`, the sample remained connected to the broker). Any particular hints on how to reproduce?

arapostol, 2021-01-21:
Did you remove all the publishing from the example? I was testing using SSL with a custom board on a GSM PPP connection.

rlubos, 2021-01-22:
Hi, yes, just a periodic ping every minute. Perhaps the issue only shows up in a more harsh environment, where some packets can get lost, I doubt that's the case for my local setup.

jukkar, 2021-01-22:
@arapostol would you be able to try this with some other connection technology than cellular?

arapostol, 2021-01-22:
I will try testing next week with Wi-Fi.

nashif, 2021-02-09:
@arapostol  did you do any additional tests?

arapostol, 2021-02-09:
My custom board with Wi-Fi hasn't arrived yet, but I think I'll will be able to test it by next week

arapostol, 2021-03-05:
Sorry for the delay, but only today I've managed to steal some time off of the development. I configured the same example to run on Wi-Fi with Zephyr 2.5 & TCP2. I will let you now of the results ASAP.

arapostol, 2021-03-06:
I was able to reproduce the disconnections when I was idling using GSM PPP, but not with Wi-Fi.

arapostol, 2021-05-18:
After testing with Ethernet I noticed the same pattern with random disconnects. It makes sense that I couldn't reproduce this with Wi-Fi as it was offloading the network connection.

closed 2021-08-03 by github-actions[bot]