zephyrproject-rtos/zephyr #31453: Add devicetree support to ieee802154_nrf5

galak, 2021-01-20:
Add DTS binding and nodes for ieee802154_nrf5

hubertmis, 2021-01-20:
I had a look at existing ieee802154 bindings, to assess what is expected. It looks that all devices described in zephyr/dts/bindings/ieee802154/ are radios with SPI interfaces. nRF5 radio does not expose a serial interface, because it is integrated as SoC.

@galak, could you share more info what kind of DTS bindings and nodes should be added regarding nRF5 802.15.4 radio?

galak, 2021-01-20:
@hubertmis since I assume the nNRF5 radio is integrated into the SoC, that a node similar to what exists for something like SPI or I2C for NRF.

Here's a rough example of what the dtsi side might look like

```
soc {
   radio@40001000 {
     compatible = "nrf,radio";
     reg = <0x40001000 0x1000>;
     interrupts = <1 NRF_DEFAULT_IRQ_PRIORITY>;
     ...
   };
};
```

galak, 2021-01-20:
@carlescufi is this someone that might help out @hubertmis with what the dts binding and node might look like for the NRF radio.

hubertmis, 2021-01-20:
It looks that such binding already exists: zephyr/dts/bindings/net/wireless/nordic,nrf-radio.yaml

galak, 2021-01-20:
> It looks that such binding already exists: zephyr/dts/bindings/net/wireless/nordic,nrf-radio.yaml

I leave it to you and @ppryga to see if that binding makes sense for the 802.15.4 driver and should get used across all the NRF SoCs that its relevant for.

galak, 2021-01-20:
The other aspect of this request is making changes to ieee802154_nrf5.c to convert `DEVICE_DEFINE` to `DEVICE_DT_DEFINE`

ppryga, 2021-01-20:
@galak and @hubertmis the binding `zephyr/dts/bindings/net/wireless/nordic,nrf-radio.yaml` was created for Direction Finding implementation. At the moment it is defined only in `dts/arm/nordic/nrf52833.dtsi` but the goal is to add it to every Nordic SOC that supports Direction Finding. It may be extended/changed to support also SOC that do not support DF.

hubertmis, 2021-01-21:
@ppryga, the same hardware peripheral is used by Direction Finding, 802.15.4, BLE. I think the only suitable solution would be to share this binding for all Nordic SoCs and extend it with options required by other radio peripheral users.

ppryga, 2021-01-21:
@hubertmis I totally agree. I'm not DTS expert. Im my imagination the best option would be to provide a generic radio node in DTS. I'm not sure how to add "extensions" to that. E.g. Direction Finding related properties that should be not allowed in SOC that do not support DF. 
Only option I see now is to make current radio node common(without DF related properties) and use it as a base for specific DTS node for SOCs that support DF.

Is there a possibility to compose DTS node from smaller features?

hubertmis, 2021-01-21:
> Only option I see now is to make current radio node common(without DF related properties) and use it as a base for specific DTS node for SOCs that support DF.

I think the way the binding is defined now is fine. Most DF properties are optional. As a first step we could add radio node to DTS of all Nordic SoCs. Later we could add property/properties needed for 802.15.4 and start using it in the 802.15.4 driver.

> Is there a possibility to compose DTS node from smaller features?

I think it is. I've seen some imports or includes. I don't know if it is needed now. Perhaps if the binding gets too big to handle, we should split it into smaller files.

ppryga, 2021-01-21:
@hubertmis Well, in my humble opinion it may be confusing for end user if she sees some DF stuff that is actually not supported for that particular SOC.
I would like to get a solution that includes DF properties only for SOCs that have DF support in radio peripheral.

hubertmis, 2021-01-21:
> it may be confusing for end user if she sees some DF stuff that is actually not supported

I agree. I understood that the only thing user see is "number of df antennas = 0", what I interpret as there is no DF (like in current nRF52833 dts). However, we can discuss if it may be somehow omitted for devices that do not support DF at all.

ppryga, 2021-01-21:
`dfe-ant-num` as well as `dfegpio#-gpios` are DF related. Those properties should be included only for SOCs that support DF.

pabigot, 2021-02-23:
I was asked in Nordic Teams to provide feedback on this.

As I understand the issue Nordic chips have a base radio, call it `nordic,nrf-radio`.  All of those support Bluetooth LE; call that `nordic,nrf-radio-ble`.  Some SoCs (e.g. nRF52833) support direction finding; call that `nordic,nrf-radio-ble-df`.  Some support 802.15.4; call that `nordic,nrf-radio-802154`.

Applications may want to support both BLE and 802.15.4 in a single application, so that'd be `nordic,nrf-radio-ble-df-802154` which is ugly, and conflates the devicetree/Kconfig distinction of describing hardware capabilities versus selecting application features.

The existing binding is primarily targeted to `nordic,nrf-radio-ble-df`: although it uses the base compatible defines DF-related properties and makes them required.  Structurally there are quite a few DF properties.

I understand that devicetree representation of 802.15.4 support needs nothing more than a boolean indicating the capability is present.

What we want is a generic Nordic devicetree node that accurately represents the radio capabilities on given SoC, while not suggesting that unsupported features can be enabled.

So, I propose the following:

* Modify the existing binding to add properties `has-dfe` and `has-802154`, as prototyped in https://github.com/pabigot/zephyr/commits/nordic/20210223a.  Remove `required` from `dfe-ant-num`: setting it to 0 to indicate "not supported" is too obscure, and apparently DFE with zero antennas has some use.

* Add a `nordic,nrf-radio` node to all Nordic SoCs that have a radio, with the two new properties set appropriately (n.b.: absence of `has-dfe;` means "not supported").  Presumably the `dfe-*` properties would be left unspecified at the SoC level, but would be defined in the board devicetree file.

* Use `BUILD_ASSERT` in the driver to diagnose cases where somebody modifies an application/board devicetree file to include DFE properties when `has-dfe` is false.  (If they also set `has-dfe`, that's their problem; we just want to help the slightly ignorant, not the determinedly obtuse.)

* Use `BUILD_ASSERT` in the driver to diagnose cases where `has-dfe` is true but no value was given to `dfe-ant-num`.

* Respect `CONFIG_IEEE802154_NRF5=n` as indicating that even when `has-802154` is true the driver should exclude that functionality, if doing so saves space.

* Add a corresponding `CONFIG_NORDIC_BT_DFE` (bikeshed) Kconfig setting to allow the same filtering of unused functionality when the SoC sets `has-dfe`, if doing so would save space and it isn't already excluded some other way.

Does that address the concerns?

hubertmis, 2021-02-23:
> Does that address the concerns?

I like this proposal. Seems the cleanest possible way.

I'm only thinking if we should connect DF with BLE, because potentially DF could be used in the future with other protocols like 802.15.4. But we can handle such mix when (if ever) it happens.

pabigot, 2021-02-23:
> I'm only thinking if we should connect DF with BLE, because potentially DF could be used in the future with other protocols like 802.15.4. But we can handle such mix when (if ever) it happens.

I'm fine with that if it's possible.  The reference manual identifies it specifically as `DFE` meaning (AFAICT) the Bluetooth LE Direction Finding Extension,  but if it's relevant to 802.15.4, certainly that can be supported.

hubertmis, 2021-02-23:
> but if it's relevant to 802.15.4, certainly that can be supported

It's not relevant now. I expect it may be in the future. I'm just guessing. I think it would be better to stick to current reference manual while we are deciding. I hope extending to other direction finding protocols, if we ever create one, will be possible with current proposal.

ppryga, 2021-02-24:
I like the proposal also.

> Add a corresponding CONFIG_NORDIC_BT_DFE (bikeshed) Kconfig setting to allow the same filtering of unused functionality when the SoC sets has-dfe, if doing so would save space and it isn't already excluded some other way.

It seems to me that new Kconfig `CONFIG_NORDIC_BT_DFE` is not needed since there is already a number of Kconfigs defined that enable the DFE handling code if it is supported by particular SOC: [Kconfig.df](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/controller/Kconfig.df)

ppryga, 2021-08-11:
This issue is partially covered by changed done by @mbolivar-nordic in this PR: https://github.com/zephyrproject-rtos/zephyr/pull/37059

fgrandel, 2022-09-18:
@hubertmis @pabigot @ppryga-nordic @galak Maybe a stupid question, but why don't you just add sub-nodes similarly to what is done on busses like uart or i2c? The main radio node could be used to describe generic radio hw aspects and capabilities like radio co-existence, antennas, etc. and the sub-nodes coud be activated separately iff ble, 802154 or any custom radio drivers should be run on that radio. This would improve encapsulation, both of the dts and the corresponding drivers, and it would be flexible to cover different socs' capabilities and for extension.