zephyrproject-rtos/zephyr #31467: samples: bluetooth: peripheral_hids: Pairing fails on the nucleo_wb55rg board.

wngur300, 2021-01-21:
**Describe the bug**
After building the peripheral_hids sample on the nucleo_wb55rg board, an error occurs when pairing is attempted.
The devices that attempted to pair were desktop (window 10), iPad Pro (4th generation), and Samsung Galaxy s21 (android).

**Environment (please complete the following information)**
OS: Linux ubuntu 18.04
Toolchain: zephyr-sdk-0.11.1
Commit SHA or Version used: 7a3b253ced7333f5c0269387a7f3ed1dee69739d

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=nucleo_wb55rg ..
3. west flash

**Expected behavior** 
It is expected that pairing should be successful if pairing is attempted on another device.

**Logs and console output**
_(Desktop Window 10 - NEXT204BT (Bluetooth CSR 4.0 Dongle) Bluetooth receiver was used.)_

1. It's a debug message after flash.
```
*** Booting Zephyr OS version 2.4.0  ***
[00:00:00.134,000] <inf> fs_nvs: 3 Sectors of 4096 bytes
[00:00:00.134,000] <inf> fs_nvs: alloc wra: 0, fd8
[00:00:00.134,000] <inf> fs_nvs: data wra: 0, 20
Bluetooth initialized
Advertising successfully started
[00:00:00.350,000] <inf> bt_hci_core: No ID address. App must call settings_load()
[00:00:00.352,000] <inf> bt_hci_core: Identity: 80:e1:26:00:41:a9 (public)
[00:00:00.352,000] <inf> bt_hci_core: HCI: version 5.0 (0x09) revision 0x0026, manufacturer 0x0030
[00:00:00.352,000] <inf> bt_hci_core: LMP: version 5.0 (0x09) subver 0x2126
```

2. This is when pairing was attempted on desktop.
```
Connected 00:1a:7d:da:71:05 (public)
Passkey for 00:1a:7d:da:71:05 (public): 441596
Security changed: 00:1a:7d:da:71:05 (public) level 4
[00:00:52.730,000] <wrn> bt_att: Ignoring unexpected request
--- 1018 messages dropped ---
[00:00:52.730,000] <err> hci_ipm: Unknown BT buf type 0
[00:00:52.730,000] <wrn> bt_att: Ignoring unexpected request
[00:00:52.730,000] <wrn> bt_att: Ignoring unexpected request
[00:00:52.730,000] <err> hci_ipm: Unknown BT buf type 0
[00:00:52.730,000] <wrn> bt_att: Ignoring unexpected request
[00:00:52.730,000] <wrn> bt_att: Ignoring unexpected request
[00:00:52.730,000] <err> hci_ipm: Unknown BT buf type 0
[00:00:52.730,000] <wrn> bt_att: Ignoring unexpected request
[00:00:52.730,000] <wrn> bt_att: Ignoring unexpected request
[00:00:52.731,000] <err> hci_ipm: Unknown BT buf type 0
[00:00:52.731,000] <wrn> bt_att: Ignoring unexpected request
[00:00:52.731,000] <wrn> bt_att: Ignoring unexpected request
[00:00:52.731,000] <err> hci_ipm: Unknown BT buf type 0
[00:00:52.731,000] <wrn> bt_att: Ignoring unexpected request
[00:00:52.731,000] <wrn> bt_att: Ignoring unexpected request
[00:00:52.731,000] <err> hci_ipm: Unknown BT buf type 0
[00:00:52.731,000] <wrn> bt_att: Ignoring unexpected request
[00:00:52.731,000] <wrn> bt_att: Ignoring unexpected request
[00:00:52.731,000] <err> hci_ipm: Unknown BT buf type 0
[00:00:52.731,000] <wrn> bt_att: Ignoring unexpected request
[00:00:52.732,000] <wrn> bt_att: Ignoring unexpected request
[00:00:52.732,000] <err> hci_ipm: Unknown BT buf type 0
[00:00:52.732,000] <err> os: ***** BUS FAULT *****
[00:00:52.732,000] <err> os:   Imprecise data bus error
[00:00:52.732,000] <err> os: r0/a1:  0x200301b4  r1/a2:  0x20032808  r2/a3:  0x00000000
[00:00:52.732,000] <err> os: r3/a4:  0x00000000 r12/ip:  0x000080bd r14/lr:  0x08003725
[00:00:52.732,000] <err> os:  xpsr:  0x6100003d
[00:00:52.732,000] <err> os: Faulting instruction address (r15/pc): 0x08003724
[00:00:52.732,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:52.732,000] <err> os: Fault during interrupt handling

[00:00:52.732,000] <err> os: Current thread: 0x20000830 (unknown)
[00:00:52.941,000] <err> os: Halting system
```
3.  I pressed reset button. You can view the warning message as below.**(Last Line)**
```
*** Booting Zephyr OS version 2.4.0  ***
[00:00:00.134,000] <inf> fs_nvs: 3 Sectors of 4096 bytes
[00:00:00.134,000] <inf> fs_nvs: alloc wra: 0, fc0
[00:00:00.134,000] <inf> fs_nvs: data wra: 0, 90
Bluetooth initialized
Advertising successfully started
[00:00:00.350,000] <inf> bt_hci_core: No ID address. App must call settings_load()
[00:00:00.353,000] <inf> bt_hci_core: Identity: 80:e1:26:00:41:a9 (public)
[00:00:00.353,000] <inf> bt_hci_core: HCI: version 5.0 (0x09) revision 0x0026, manufacturer 0x0030
[00:00:00.353,000] <inf> bt_hci_core: LMP: version 5.0 (0x09) subver 0x2126
[00:00:00.358,000] <wrn> bt_hci_core: Set privacy mode command is not supported
```

4. You can't discover a device on your desktop, and you can search it on your smartphone, but when you try to connect, you'll see the following message:
```
Connected 4b:56:cf:36:4c:44 (random)
Failed to set security
Security failed: 4b:56:cf:36:4c:44 (random) level 1 err 8
[00:03:52.408,000] <wrn> bt_att: Unhandled ATT code 0x1d
Disconnected from 4b:56:cf:36:4c:44 (random) (reason 0x13)
```
**Additional context**
The same is true of iPad Pro (4th generation) and Samsung Galaxy s21 (android).

What should I do?
I'm not sure what's wrong.
What do you think is the problem?
Thank you for reading.

str4t0m, 2021-01-21:
Hi,
this sample works fine for me on the nucleo_wb55rg.
Are you using the full stack or the HCILayer Coprocessor Binary?
Have you already tried to update the Coprocessor Binary to the latest version?

wngur300, 2021-01-22:
> Hi,
> this sample works fine for me on the nucleo_wb55rg.
> Are you using the full stack or the HCILayer Coprocessor Binary?
> Have you already tried to update the Coprocessor Binary to the latest version?

As you said, after upgrading **stm32wb5x_FUS_fw_1_0_2.bin**, I updated **stm32wb5x_BLE_Stack_full_fw.bin**.
Pairing works well on desktop (window10).
Thank you!

However, if I disconnect pairing from desktop and try pairing from other devices such as iPad Pro (4th generation), Samsung Galaxy s21 (android), it will not connect.

**When attempting to connect from iPad Pro (4th generation), the following log message occurs:**
```
Connected 48:43:87:77:d5:c7 (random)
Failed to set security
Security failed: 48:43:87:77:d5:c7 (random) level 1 err 8
Disconnected from 48:43:87:77:d5:c7 (random) (reason 0x13)
```

**When attempting to connect from Samsung Galaxy s21 (android), the following log message occurs:**
```
Connected 61:65:42:5a:cc:2f (random)
Failed to set security
Security failed: 61:65:42:5a:cc:2f (random) level 1 err 8
[00:07:30.478,000] <wrn> bt_att: Unhandled ATT code 0x1d
Disconnected from 61:65:42:5a:cc:2f (random) (reason 0x13)
```
Can you pair well on other devices?
What do you think is the problem?

erwango, 2021-01-25:
I'm seeing the same on `reel_board` using a Samsung Galaxy A51:
```
Passkey for 69:40:A5:A2:5D:67 (random): 143238                                  
Security failed: 69:40:A5:A2:5D:67 (random) level 1 err 1                       
Disconnected from 69:40:A5:A2:5D:67 (random) (reason 0x13)
```
So I guess this is more related to sample implementation than the target. @jhedberg can you have a look ?

jhedberg, 2021-01-25:
@erwango can you provide HCI logs for the failure?

joerchan, 2021-01-25:
@erwango You have a different error (authentication failure) than what is posted in the original bug description (unspecified).
Maybe you put in the wrong passkey?

@wngur300 I think your issue here (https://github.com/zephyrproject-rtos/zephyr/issues/31467#issuecomment-765097635) is related to the number of supported pairings, you can increase this trough the kconfig option CONFIG_BT_MAX_PAIRED.

carlescufi, 2021-01-28:
@erwango and @wngur300 could you please reply to [this comment](https://github.com/zephyrproject-rtos/zephyr/issues/31467#issuecomment-766734327)

erwango, 2021-01-28:
@joerchan @carlescufi:
Using `CONFIG_BT_MAX_PAIRED=3`, I've been able to connect 2 different windows PC (Using `2` didn't help).
And I'm observing strange behavior with my smartphone (pairing randomly failed), but that doesn't seem related to Zephyr stack (when pairing fails on smartphone, I can pair on laptop).
So from my point of view, this is working fine.

wngur300, 2021-01-29:
> @erwango and @wngur300 could you please reply to [this comment](https://github.com/zephyrproject-rtos/zephyr/issues/31467#issuecomment-766734327)

There was a problem with the nucleo_wb55 board, so I bought a new board and did other work, so I was late to reply.
I'm sorry.
As @erwango said, I changed the value of CONFIG_BT_MAX_PAIRED and it worked well on the nrf52840dk_nrf52840, and I will test and answer on the nuleo_wb55rg board.

wngur300, 2021-01-29:
> @joerchan @carlescufi:
> Using `CONFIG_BT_MAX_PAIRED=3`, I've been able to connect 2 different windows PC (Using `2` didn't help).
> And I'm observing strange behavior with my smartphone (pairing [...]

That's right. I changed the value of CONFIG_BT_MAX_PAIRED = 3 and the problem was solved.
Thank you!

erwango, 2021-01-29:
@joerchan If `CONFIG_BT_MAX_PAIRED = 3` is the expected setting in this case, then we can close the issue.

joerchan, 2021-02-02:
Great. If the application is to be paired with multiple peers then the MAX_PAIRED needs to be increased to support this.
When flashing the board the bond storage might remain in flash (use `west flash --erase`  to delete entire flash).

The sample is using the default of 1 connection and 1 bond, I don't see a reason to change this, so I'll close this issue as everything seems to be resolved.

closed 2021-02-02 by joerchan