zephyrproject-rtos/zephyr #31473: Failed phy request not retried and may prevent DLE procedure during auto-initiation

mtpr-ot, 2021-01-21:
When an auto-initiated phy request collides internally with other active LLCP, the request has been marked as done, and no continuation of auto-initiation is triggered.
In a similar case where an auto-initiated phy request collides with other procedure initiated by the peer, the "Different Transaction Collision" error code is ignored, and auto-initiation continued, effectively skipping phy update.

In the first case, phy is not updated correctly and the next procedure is not initiated, leaving both phy and data length at unexpected (default) values. In the second case, only problem is the phy not updated.

**To Reproduce**
1) After connection is established, the central/peripheral LLCP timing must be such that the auto-initiated phy update request from peripheral is sent just **after** e.g. a encryption request is received by the peripheral. This will trigger a "disallowed" result from call to `ll_phy_req_send`

2) After connection is established, the central/peripheral LLCP timing must be such that the auto-initiated phy update request from peripheral is sent just **before** e.g. a encryption request is received by the peripheral. This will get the request in the air, and central responds with a rejection: "Different Transaction Collision".

**Expected behavior**
The host app expects that at least a reasonable effort has gone into executing the auto-initiated LLCPs, doing simple retries where applicable. It would be expected that no matter what, a failing auto-initiated phy request would not prevent DLE.

**Impact**
Link is left in 1 Mbps phy when 2 is expected, and even worse, data transfers may be performed using default size (27 byte) DL.

**Environment**
Issue is seen on vendor implementation based on Zephyr 2.2.1

mtpr-ot, 2021-01-26:
Updated description to only consider "Different Transaction Collison" issue, as the "DISALLOWED" issue was fixed.

carlescufi, 2021-01-28:
Downgrading to low because this has a very simple workaround, which is to retrigger the procedure from the application, and there are no explicit guarantees for retrial in the API.

closed 2021-09-17 by mtpr-ot