zephyrproject-rtos/zephyr #31475: TCP keepalive

avigreen1978, 2021-01-21:
**Is your feature request related to a problem? Please describe.**
The scenario is as follows:
We have a TCP connection over wifi interface.
Then, the peer abruptly closes wifi connection, and FIN is not received on zephyr side.
Because FIN is not received, the connection will never close, even though the other side is disconnected.

**Describe the solution you'd like**
posix setsockopt has an SO_KEEPALIVE option, and then tcp sends keep-alive when there's no traffic.

jukkar, 2021-01-21:
The keep-alive option is indeed not implemented in TCP implementation. This would be a good to have feature so setting labels accordingly.

ycsin, 2021-08-30:
Is this the same as #34044 ? @jukkar @rlubos

rlubos, 2021-09-06:
> Is this the same as #34044 ? @jukkar @rlubos @mniestroj

Yes, the other one seems like a duplicate. I propose to close this however since #34044 already has a linked PR.

closed 2021-09-06 by rlubos