zephyrproject-rtos/zephyr #31488: build failure w/twister and SDK 0.12.1 related to

galak, 2021-01-21:
Several tests fail to build on CC13xx/CC26xx platforms related to power management:

```
tests/kernel/tickless/tickless_concept/kernel.tickless.concept in cc1352r1_launchxl:tests/kernel/tickless/tickless_concept/kernel.tickless
tests/kernel/timer/timer_api/kernel.timer.tickless in /workdir/zephyr/twister-out/cc1352r1_launchxl/tests/kernel/tickless/tickless_concept/kernel.tickless.concept:tests/kernel/timer/timer_api/kernel.timer
tests/kernel/tickless/tickless_concept/kernel.tickless.concept in cc1352r_sensortag:tests/kernel/tickless/tickless_concept/kernel.tickless
tests/kernel/timer/timer_api/kernel.timer.tickless in /workdir/zephyr/twister-out/cc1352r_sensortag/tests/kernel/tickless/tickless_concept/kernel.tickless.concept:tests/kernel/timer/timer_api/kernel.timer
tests/kernel/tickless/tickless_concept/kernel.tickless.concept in cc26x2r1_launchxl:tests/kernel/tickless/tickless_concept/kernel.tickless
tests/kernel/timer/timer_api/kernel.timer.tickless in /workdir/zephyr/twister-out/cc26x2r1_launchxl/tests/kernel/tickless/tickless_concept/kernel.tickless.concept:tests/kernel/timer/timer_api/kernel.timer
```

The failure is some form of:

```
/workdir/zephyr/subsys/power/pm_ctrl.c: In function 'pm_ctrl_disable_state':
/workdir/zephyr/subsys/power/pm_ctrl.c:24:6: error: array subscript  is outside array bounds of 'atomic_t[0]' {aka 'int[]'} [-Werror=array-bounds]
   24 |  v = atomic_inc(&power_state_disable_count[state]);
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/workdir/zephyr/subsys/power/pm_ctrl.c:17:17: note: while referencing 'power_state_disable_count'
   17 | static atomic_t power_state_disable_count[POWER_STATE_MAX];
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~
/workdir/zephyr/subsys/power/pm_ctrl.c: In function 'pm_ctrl_enable_state':
/workdir/zephyr/subsys/power/pm_ctrl.c:36:6: error: array subscript  is outside array bounds of 'atomic_t[0]' {aka 'int[]'} [-Werror=array-bounds]
   36 |  v = atomic_dec(&power_state_disable_count[state]);
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/workdir/zephyr/subsys/power/pm_ctrl.c:17:17: note: while referencing 'power_state_disable_count'
   17 | static atomic_t power_state_disable_count[POWER_STATE_MAX];
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~
/workdir/zephyr/subsys/power/pm_ctrl.c: In function 'pm_ctrl_is_state_enabled':
/workdir/zephyr/subsys/power/pm_ctrl.c:47:21: error: array subscript  is outside array bounds of 'atomic_t[0]' {aka 'int[]'} [-Werror=array-bounds]
   47 |  return (atomic_get(&power_state_disable_count[state]) == 0);
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/workdir/zephyr/subsys/power/pm_ctrl.c:17:17: note: while referencing 'power_state_disable_count'
   17 | static atomic_t power_state_disable_count[POWER_STATE_MAX];
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
```

galak, 2021-01-21:
At glancing at the code, the assumption that POWER_STATE_MAX can be used to initialize and array to track information seems a bit invalid, since the enum has negative values like `POWER_STATE_AUTO` and `POWER_STATE_ACTIVE` there for the value of POWER_STATE_MAX doesn't seem to be the number of valid power states.

ceolin, 2021-01-21:
The problem is that depend on build configuration `POWER_STATE_MAX` can be 0. As the check in the API is in an assert the compiler is warning about a possible invalid access to the array.
https://github.com/zephyrproject-rtos/zephyr/pull/31273 fixes this.

closed 2021-01-22 by nashif