zephyrproject-rtos/zephyr #31503: drivers: i2c: i2c_nrfx_twim  Power Consumption rises after I2C data transfer

YoYo0304, 2021-01-22:
**Describe the bug**
We are using a nrf52832 based board. There are some I2C devices like tsm12 and eeprom.
The power consumption is 200ua when the I2C buses of the nrf52832 is enabled, but without data transfering. 
But when data transfer throw the I2C bus, the power consumption of the board increased by about 400ua.
The power consumption will never fall back again, even when the data transfering is over.

What have you tried to diagnose or workaround this issue?
1. When the issue is taking place, i have tried to disconnect the physical connection of all the i2c devices 
    included the tsm12 and the eeprom.  But it is useless. So the increase of the power consumption issue is caused 
   by the chip of nrf52832.
2. But the i2c driver of nrf52832 has disabled the i2c bus after data transfering by calling this function: 
   "nrfx_twim_disable(&get_dev_config(dev)->twim);" . So it seems that the power consumption issue is not caused by
   the i2c bus of the nrf52832. 
3. Then i tried to disable  all the interupts of the i2c bus of the nrf52832. The issue does not take place.
4. I tried to delete the calling of the function : k_sem_take(&(get_dev_data(dev)->completion_sync),
    I2C_TRANSFER_TIMEOUT_MSEC).  The issue does not take place.
5. So it seems that the interupts conflict with this function：k_sem_take. **Maybe the interupts affect the power saving 
    strategy of the kernel**.

**To Reproduce**
Steps to reproduce the behavior:
1. Find a nrf52 based board with some i2c devices
2. run the sample
3. Test the power cosumpertion

YoYo0304, 2021-01-22:
A nrf52832 based board without a i2c device is also can be used to reproduce the issue. Becase the errors can also trigger 
an interupt.

nashif, 2021-01-23:
@carlescufi  please assign

dueringa, 2021-02-19:
@YoYo0304 A quick info from me, who has been developing on the nrf52832 - there's an errata 89 (https://infocenter.nordicsemi.com/index.jsp?topic=%2Ferrata_nRF52832_EngB%2FERR%2FnRF52832%2FEngineeringB%2Flatest%2Fanomaly_832_89.html) - power cycling the TWIM/SPIM register *should* help. 

However, in doing so, the peripheral loses its configuration, and currently, there seems to be no way to re-initialize a device driver. From what I've gathered in Slack, there's an issue where this is being worked on.

mniestroj, 2021-02-19:
There are at least two ways to workaround increased power consumption:
1) follow nrf52832 errata, as @dueringa mentioned; this means that i2c peripheral needs to be reinitialized before each i2c transfer, but such behavior needs changes in nrf i2c driver,
2) stop using GPIOTE; there is an open PR (#31823) to "emulate" edge interrupts using other GPIO capabilities in nRF.

anangl, 2021-04-27:
@YoYo0304 can you try to switch from TWIM to TWI, i.e. set `compatible = "nordic,nrf-twi";` instead of `compatible = "nordic,nrf-twim";` in the devicetree node for the instance you are using?

carlescufi, 2021-05-11:
Closing due to lack of response. Please reopen if this is still an issue after you have tried @anangl's suggestion.