zephyrproject-rtos/zephyr #31517: UP²: no console output anymore (git bisect findings inside)

tbursztyka, 2021-01-22:
**Describe the bug**
Any samples running on UP² that has a console output should work, but it does not anymore.

**To Reproduce**

I tried the tests/arch/x86/info sample
up_squared board  (so x86_64) on multiboot (zephyr.strip)

**Expected behavior**
A lot of output on the multiboot, acpi and RTC

**Impact**
I cannot test my work on UP² atm, and I am using it extensively

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain:Zephyr SDK
 - Commit SHA or Version used

**Additional context**

I ran a git bisect, and it finds the commit 3660040e22042e0e67005da0a7229fc224e089d0 to be the first bad one.

tbursztyka, 2021-01-22:
@andrewboie @nashif @dcpleung @andyross  Please have a look

@jhedberg Is the console still working properly on EHL on latest master? Perhaps, the issue could affect more that just the UP².

jhedberg, 2021-01-22:
@andrewboie @tbursztyka I just discovered that the same commit 3660040 also breaks zephyr.efi creation (the resulting file is too small (some 86k) and not a valid EFI binary according to EFI BIOS. @andyross FYI

finikorg, 2021-01-22:
Cannot boot any of EHLs

jhedberg, 2021-01-22:
@jenmwms FYI

nashif, 2021-01-22:
> Cannot boot any of EHLs, commit that broke seems to be [513b9f5](https://github.com/zephyrproject-rtos/zephyr/commit/513b9f598fc1135219d555013e8aa821447662d9)

this most likely would be a different issue, unrelated to paging

nashif, 2021-01-22:
paging commits reverted now, wrt to EFI, this should be filed as its own bug if this is really a problem.

tbursztyka, 2021-01-25:
EHL issue was definitely another one. 

UP² issue is still there.

nashif, 2021-01-25:
duplicate of #31566

PR #31540 (2021-01-22): x86: Fix zefi.py creating valid images
When zefi.py was changed to pass compiler and objcopy the flag to
objcopy for the EFI target was dropped.  This is because the current
SDK (0.12.1) doesn't support that target type for objcopy.  However,
target is necessary for the images to be created correctly and boot.

Switch back to use the host objcopy as a stop gap fix, until the SDK
can support target for EFI.

Fixes: #31517
commit 895277f9 (2021-01-22): x86: Fix zefi.py creating valid images

When zefi.py was changed to pass compiler and objcopy the flag to
objcopy for the EFI target was dropped.  This is because the current
SDK (0.12.1) doesn't support that target type for objcopy.  However,
target is necessary for the images to be created correctly and boot.

Switch back to use the host objcopy as a stop gap fix, until the SDK
can support target for EFI.

Fixes: #31517
closed 2021-01-25 by nashif