zephyrproject-rtos/zephyr #31518: Communication across Memory Domains (IPC)

Sasha-Spartak, 2021-01-22:
# Motivation
In the modern world security is a critical part of nearly every system design and there are two fundamental security principles: Least Privilege and Separation/Isolation.
Zephyr supports these principles by enabling multiple isolated user-level memory domains.
Design of a system with multiple isolated user-level memory domains requires communication mechanism between the memory domains. Such mechanism is frequently referred as IPC (Inter-Process Communication).
This request proposes to introduce IPC which:
 - Allows secure communication between user-level memory domains
 - Allows to move some drivers from Kernel to User privilege level
 - Supports embedded systems with limited resources (memory)
 - Facilitates transition from a monolithic user-level application design to set of multiple isolated user-level components

Presence of such IPC would make Zephyr security offering more complete and as a result would make Zephyr more attractive.
# Terminology
**IPC server (or server)**
IPC Server is a thread receiving requests/events from IPC clients, processing them, and (if needed) sending responses to the clients.
**IPC client (or client)**
IPC client is a thread sending requests/events to IPC server and (if needed) receiving responses from it.
**IPC Channel**
IPC Channel is a kernel object allowing communication between IPC clients and IPC server
**Queue**
An abstract data type allowing to organize multiple items in a particular order from their delivery perspective. Queue can be implemented in different ways (array, linked list, etc.).
**IPC request (or request)**
A method allowing IPC client to either ask IPC server to provide some service (IPC call) or notify IPC server about occurence of some event (IPC Event). 
**IPC call**
Type of IPC request used in the cases where IPC client waits for a response from IPC server
**IPC Event**
    Type of IPC request used in the cases where IPC client just notifies IPC server about some event without being blocked. An IPC Event is classified as either IPC_EVENT or IPC_INTERRUPT. An event is classified as IPC_INTERRUPT where Kernel needs to notify user-level driver (IPC Server) about occurrence of an interrupt. IPC_EVENT classification is used in all other cases. IPC Events classified as IPC_INTERRUPT are always delivered before the ones classified as IPC_EVENT.
# IPC Overview
IPC is a “many-to-one” communication mechanism (multiple clients and single server) where the communication is always initiated by the clients. The clients can communicate with the server in two ways: Synchronously and Asynchronously.
Synchronous IPC is designed to support “procedure call” paradigm where the client “calls” a function implemented by the server which is executed in the server’s thread context. Execution of the client's thread is blocked until response from the server is received.
Asynchronous IPC is designed to allows IPC clients to notify IPC server about different events. These notifications are delivered to the server without blocking execution of the client's thread.
There are two types of IPC clients: Privileged and Non-Privileged. The only difference between Privileged and Non-Privileged clients is that the privileged ones have more control over priority and classification of their requests.
All IPC communications go through a kernel object called IPC Channel. There could be multiple IPC Channels where each of them used by one specific IPC server and particular set of IPC clients. Note that a thread playing IPC server role for one IPC Channel may be IPC client for other IPC Channel.
Access permissions to IPC Channel are granted to clients and servers at the build/design time. IPC server and clients have different permissions in accessing same IPC Channel (part of IPC APIs is available for IPC servers only and other part of IPC APIs is available for IPC clients only).
IPC Channels are initialized as part of kernel initialization and are ready for use from very beginning of every server/client thread execution.
The pictures below show different types of client-server interactions via IPC Channel and possible system design based on IPC.
[image]

[image]
# IPC APIs (description)
There are four APIs associated with IPC Channel. Details description of the APIs will be given later.
**ipc_call**
This API is used by IPC clients for synchronous IPC with IPC server. It allows the clients to pass to the server a request (IPC call) with additional data and receive a response with additional data (similar to a function call with input and output parameters).
**ipc_event**
This API is used by IPC clients for asynchronous communication with IPC server. It allows the clients to notify the server about particular event and pass limited (by the server design) amount of the data associated with the event. Entire request (including the data) is copied to Event Queue of IPC Channel without blocking client's thread execution.
**ipc_get_request**
This API is used by IPC servers for receiving the requests sent via “ipc_call” and “ipc_event” APIs. This API is blocking meaning that execution of the server’s thread is suspended until either a request/event is delivered or the timeout expires.
**ipc_response**
This API is used by IPC server in order to respond to the client’s requests sent via “ipc_call” APIs. Calling this API does not block execution of the server’s thread.
# IPC Channel
IPC Channel is a composite Kernel object including Call Queue, Event Queue, and Channel State information.
**Call Queue**
Queue object likely implemented as a linked list. Elements in the queue represent IPC client threads waiting in "ipc_call" API. Depending of "ipc_call" parameter the elements are added either according to the client's thread priority or to the end.
**Event Queue**
Queue object where every element is a complete IPC request made through "ipc_event" API. Although each element tracks priority of the thread which called "ipc_event" API, order of the elements in the queue depends on the parameters used in "ipc_event" API. IPC events classified as IPC_INTERRUPT are always server first.
**Channel State**
Data structure containing all the information required for the proper functioning of IPC Channel. Exact structure of Channel State is very implementation dependent, but in this or other form it keeps the the following information and control data:
- Server Thread
Awareness of the server thread associated with the given IPC Channel may be needed for several purposes: a) Scheduling decisions where IPC request release the server thread waiting in "ipc_get_request" API, b) Access control to distinguish between IPC server and IPC client allowed to access the given IPC Channel.
- Call State
State of the served IPC clients in Call Queue (e.g. waiting to send request vs. waiting to get response). It may be used checking correctness of the server behavior (e.g. making sure that the server responds to the client that already "sent" the request and waits for response).
- Request delivery QoS
IPC Channel should provide some fairness in delivering the requests (IPC calls and IPC events) to the server. There shouldn't bea situation that a client (except privileged one) sending IPC events completely blocks delivery of IPC call requests (and vice versa).
- Event sending QoS
IPC design should avoid any dynamic allocation of the resources. Thus capacity of Event Queue is fixed (set and design time). In this case one IPC client could fill Event Queue with its requests and block ability of other clients to send events. IPC Channel need to implement a mechanism preventing such situation (e.g. maintain for each IPC client maximal number of enqueued IPC events). 
# Thread Scheduling
Synchronous IPC can be seen as a mechanism allowing to "move" execution of particular function from the context of IPC client thread to the context of IPC server thread. All the functions executed by IPC client thread are executed at the priority assigned to IPC client thread, but once the function execution moves to the context of IPC server thread the priority may change. In IPC client's flow designed to be executed at low priority some part would be executed at higher priroity (affecting scheduling of other threads) or in IPC client's flow designed to be executed at high priority some part would be executed at lower priroity (potentially leading to QoS degradation). As a mitigation for the issues related to the changes of schedulng priority IPC implements "priority inheri [... cut ...]
This API is used by IPC server for sending response to the client’s requests issued via “ipc_call” APIs. Calling this API does not block execution of the server’s thread.
Parameters:
- **channel** - reference of the IPC Channel (kernel object ID, pointer, handle, ...)
- **client_id** - ID of the Client the response should be sent to.
- **out_data** - Pointer to the buffer containing the data to be sent to the client. This parameter is ignored when “out_buf_size” parameter is 0.
- **out_data_size** - size in bytes of the data in the buffer referred by “out_data” parameter. This parameter can be 0 if there is no output data.

dcpleung, 2023-04-15:
Have you looked at zbus? It is a message bus for Zephyr.

aangarov, 2023-04-16:
No, I didn't look at zbus. Unfortunately I had to switch to other activities and do not work anymore on Zephyr.

rodrigopex, 2023-05-18:
@aangarov, Could we close this issue so far?

BorisKrichevski, 2023-05-30:
Sorry for late response. 
@aangarov moved to other activities and is not involved into Zephyr related project
I cover for him and need couple weeks to learn current status and update.

rodrigopex, 2023-06-13:
@BorisKrichevski, any update on that?

BorisKrichevski, 2023-06-18:
Hi Rodrigo

Sorry for delay
We can withdraw the issue of “Communication across Memory Domains” in the form that Alexander proposed it.
Our Zephyr software should keep maximal compatibility with Linux software we develop for parallel project with unlimited memory (DDR) and extended  capabilities.
End last week we finally decided to use gRPC for communication between Linux processes.
Zephyr does not have gRPC support, we need to choose/develop/modify protocol for communication between Memory Domains that provides maximal compatibility with gRPC with minimal memory usage.
Advices are appreciated.

Boris Krichevski
NCNG FW Architecture

Intel Israel (74) Limited

This e-mail and any attachments may contain confidential material for
the sole use of the intended recipient(s). Any review or distribution
by others is strictly prohibited. If you are not the intended
recipient, please contact the sender and delete all copies.

rodrigopex, 2023-06-20:
@BorisKrichevski, thank you. I am closing the issue. If you need to discuss something related to that, please reopen it.

closed 2023-06-20 by rodrigopex