zephyrproject-rtos/zephyr #31524: littlefs: Too small heap for file cache.

r2r0, 2021-01-22:
**Describe the bug**
fs_open for CONFIG_FS_LITTLEFS_NUM_FILESth file returns -ENOMEM.
I.e. CONFIG_FS_LITTLEFS_NUM_FILES=3 and when fs_open is caled for 3rd file it returns -ENOMEM.
After incrementing CONFIG_FS_LITTLEFS_NUM_FILES and opening only CONFIG_FS_LITTLEFS_NUM_FILES-1 files problem is not present.
Even after increasing file_cache_pool (in littlefs_fs.c) by 128 bytes problem is not present.

**To Reproduce**
Try to open at the same time CONFIG_FS_LITTLEFS_NUM_FILES number of files on littleFS.

**Expected behavior**
No error from fs_open.

**Impact**
annoyance

**Logs and console output**
None
**Environment (please complete the following information):**
 - OS: Linux
 - ToolchainZephyr SDK 0.12.1
 - Commit SHA c31ce55c583a3f68a6d4b815b1693d83098e2ab3

**Additional context**
Maybe commit [fcd392f](https://github.com/zephyrproject-rtos/zephyr/commit/fcd392f6ceff5d552d52b3337f373ed8ed14a30e#diff-c288df0c39e91d11f900c71b4978331560d72d79faf18fef58c65531f420a914) caused this problem because heap size available to user != declared size of heap.

pabigot, 2021-01-26:
#31597 may also be relevant.

pabigot, 2021-01-28:
I am unable to reproduce this problem.  Using an augmented test in https://github.com/pabigot/zephyr/commits/repro/31524 I can open the required number of files, and directories, on `nrf52840dk_nrf52840` without error.  (Below this was done with the default count; reconfiguring to 3 also worked.)

```
CONFIG_FS_LITTLEFS_NUM_FILES=4                                                  
opening /sml/A                                                                  
opening /sml/B                                                                  
opening /sml/C                                                                  
opening /sml/D                                                                  
Close and unlink /sml/D                                                         
Close and unlink /sml/C                                                         
Close and unlink /sml/B                                                         
Close and unlink /sml/A                                                         
CONFIG_FS_LITTLEFS_NUM_DIRS=4                                                   
making and opening directory /sml/DA                                            
making and opening directory /sml/DB                                            
making and opening directory /sml/DC                                            
making and opening directory /sml/DD                                            
Close and rmdir /sml/DD                                                         
Close and rmdir /sml/DC                                                         
Close and rmdir /sml/DB                                                         
Close and rmdir /sml/DA                                                         
unmounting /sml
```

@r2r0 could you please check this on your configuration?  It may be specific to a particular type of flash or platform.

r2r0, 2021-01-29:
Attached test fails on QEMU x86 and similar test (the same code and the same LITTLEFS config) also fails on our board with STM32WB55 (external FLASH over QSPI).
[littlefs_max_files.zip](https://github.com/zephyrproject-rtos/zephyr/files/5893013/littlefs_max_files.zip)

pabigot, 2021-01-29:
@r2r0 thanks, that gave me the clue.  Please verify solution in #31781.

r2r0, 2021-01-29:
@pabigot Thanks for quick reply.
It works but it looks for me a little weird because I expected that littleFS will allocate (in file_cache_pool): 
CONFIG_FS_LITTLEFS_NUM_FILES * CONFIG_FS_LITTLEFS_CACHE_SIZE bytes 
but now it allocates: 
CONFIG_FS_LITTLEFS_NUM_FILES * CONFIG_FS_LITTLEFS_NUM_FILES * CONFIG_FS_LITTLEFS_CACHE_SIZE bytes
Did I miss something?

pabigot, 2021-01-29:
> Did I miss something?

No, apparently I did.  I'll take another look later.

pabigot, 2021-01-29:
So here's what's going on.

Originally the file cache used a mem_pool.  With that data structure it appears metadata associated with allocations was held separately, so that the entire requested storage space was available for application use.

In #28611 the `k_mem_pool` API was deprecated, and existing uses switched to a heap implementation.  It appears in this implementation metadata associated with allocations is taken from the pool itself, which breaks the expectation that you can reserve N blocks of size B and actually allocate N blocks of size B.

I've confirmed this by verifying that the test program above passes at commit 78614bf271c2669912730120fddb32e0102710e0 but breaks at commit fcd392f6ceff5d552d52b3337f373ed8ed14a30 where the deprecated API was replaced by a heap.

The Kconfig options used to allocate the space for per-file caches are completely obsoleted by the change in data structure.

What I've done in #31781 is deprecate the old mem_pool Kconfig options, and add a new option that provides more direct control of the size of the memory region to allocate file caches.  The chain of defaults producing that size have been updated to include the required slop so that allocating N blocks of size B works again.

r2r0, 2021-02-01:
So, my guessing about problem reason was correct, nice.
Thanks @pabigot , it works very well.

pabigot, 2021-02-01:
Let's leave this open until the fix is merged.

PR #31781 (2021-02-02): fs: littlefs: fix default cache size
Originally the file cache used a mem_pool, but that data structure has been deprecated and replaced by a heap that includes metadata in the  heap area.  As a result attempts to allocate all blocks will fail because some of the reservation intended for cache data is now holding metadata instead.
    
It's not immediately clear how to adjust the required heap size to support this metadata as it depends on heap chunk units and data structures that are not visible to the application.  Experimentally a value of 24 bytes works, while smaller values do not.
    
F [... cut]
closed 2021-02-02 by nashif