zephyrproject-rtos/zephyr #31531: STM32 can driver don't set prescaler

AlPon1969, 2021-01-22:
In function "can_stm32_set_timing" on file can_stm32.c the set of BRP value is missing.
To solve this is necessary to change BTR register load from:

can->BTR |= (((timing->phase_seg1 - 1) & 0x0F) << CAN_BTR_TS1_Pos) |
		    (((timing->phase_seg2 - 1) & 0x07) << CAN_BTR_TS2_Pos) |
		    (((timing->sjw        - 1) & 0x07) << CAN_BTR_SJW_Pos);

to

	can->BTR |= (((timing->phase_seg1 - 1) & 0x0F) << CAN_BTR_TS1_Pos) |
		    (((timing->phase_seg2 - 1) & 0x07) << CAN_BTR_TS2_Pos) |
		    (((timing->sjw        - 1) & 0x07) << CAN_BTR_SJW_Pos) |
		    (((timing->prescaler  - 1) & 0x1F) << CAN_BTR_BRP_Pos);

alexanderwachter, 2021-01-23:
Hi, this problem is already solved. Pleas pull the latest master and check if your problem still exist.

alexanderwachter, 2021-01-26:
Closing because it got fixed.
@AlPon1969 if you still see this issue, please reopen.

closed 2021-01-26 by alexanderwachter