zephyrproject-rtos/zephyr #31549: tests/kernel/lifo/lifo_usage/kernel.lifo.usage fails on m2gl025_miv

nashif, 2021-01-23:
```
23:07:07.9621 [WARNING] plic: Unhandled write to offset 0x200000, value 0x0.
23:07:07.9675 [WARNING] uart: Unhandled write to offset 0xC. Unhandled bits: [0] when writing value 0x1. Tags: BIT8 (0x1).
23:07:07.9819 [DEBUG] uart: [+0.38s host +0.1ms virt 0.1ms virt from start] *** Booting Zephyr OS build zephyr-v2.4.0-3530-g53a00b41ce24  ***
23:07:07.9836 [DEBUG] uart: [+1.65ms host +0.1ms virt 0.2ms virt from start]   Running test suite test_lifo_usage
23:07:07.9840 [DEBUG] uart: [+0.5ms host +0s virt 0.2ms virt from start]   ===================================================================
23:07:07.9843 [DEBUG] uart: [+0.26ms host +0s virt 0.2ms virt from start]   START - test_lifo_nowait
23:07:07.9904 [DEBUG] uart: [+6.09ms host +0.1ms virt 0.3ms virt from start]    PASS - test_lifo_nowait
23:07:07.9909 [DEBUG] uart: [+0.53ms host +0s virt 0.3ms virt from start]   ===================================================================
23:07:07.9914 [DEBUG] uart: [+0.52ms host +0.1ms virt 0.4ms virt from start]   START - test_lifo_wait
23:07:07.9920 [DEBUG] uart: [+0.62ms host +0s virt 0.4ms virt from start]    PASS - test_lifo_wait
23:07:07.9926 [DEBUG] uart: [+0.62ms host +0.1ms virt 0.5ms virt from start]   ===================================================================
23:07:07.9929 [DEBUG] uart: [+0.26ms host +0s virt 0.5ms virt from start]   START - test_timeout_empty_lifo
23:07:07.9953 [DEBUG] uart: [+2.39ms host +0.2ms virt 0.7ms virt from start]    PASS - test_timeout_empty_lifo
23:07:07.9958 [DEBUG] uart: [+0.49ms host +0s virt 0.7ms virt from start]   ===================================================================
23:07:07.9961 [DEBUG] uart: [+0.36ms host +0.1ms virt 0.8ms virt from start]   START - test_timeout_non_empty_lifo
23:07:07.9967 [DEBUG] uart: [+0.6ms host +0s virt 0.8ms virt from start]    PASS - test_timeout_non_empty_lifo
23:07:07.9974 [DEBUG] uart: [+0.66ms host +0.1ms virt 0.9ms virt from start]   ===================================================================
23:07:07.9976 [DEBUG] uart: [+0.26ms host +0s virt 0.9ms virt from start]   START - test_timeout_lifo_thread
23:07:08.0015 [DEBUG] uart: [+3.81ms host +0.3ms virt 1.2ms virt from start]    PASS - test_timeout_lifo_thread
23:07:08.0021 [DEBUG] uart: [+0.6ms host +0.1ms virt 1.3ms virt from start]   ===================================================================
23:07:08.0024 [DEBUG] uart: [+0.32ms host +0s virt 1.3ms virt from start]   START - test_timeout_threads_pend_on_lifo
23:07:08.0037 [DEBUG] uart: [+1.26ms host +0.1ms virt 1.4ms virt from start]    thread (q order: 2, t/o: 0, lifo 0x8004099c)
23:07:08.0043 [DEBUG] uart: [+0.6ms host +0.2ms virt 1.6ms virt from start]
23:07:08.0057 [DEBUG] uart: [+1.4ms host +0.1ms virt 1.7ms virt from start]       Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/lifo/lifo_usage/src/main.c:140: test_multiple_threads_pending: (data->timeout_order not equal to ii)
23:07:08.0059 [DEBUG] uart: [+0.29ms host +0s virt 1.7ms virt from start]    *** thread 2 woke up, expected 1
23:07:08.0060 [DEBUG] uart: [+39µs host +0s virt 1.7ms virt from start]
23:07:08.0068 [DEBUG] uart: [+0.83ms host +0.1ms virt 1.8ms virt from start]    FAIL - test_timeout_threads_pend_on_lifo
23:07:08.0074 [DEBUG] uart: [+0.59ms host +0.1ms virt 1.9ms virt from start]   ===================================================================
23:07:08.0077 [DEBUG] uart: [+0.3ms host +0s virt 1.9ms virt from start]   Test suite test_lifo_usage failed.
23:07:08.0088 [DEBUG] uart: [+1.05ms host +0s virt 1.9ms virt from start]   ===================================================================
23:07:08.0091 [DEBUG] uart: [+0.34ms host +0.1ms virt 2ms virt from start]   PROJECT EXECUTION FAILED
```

nashif, 2021-01-27:
@andyross this is the one I was referring to

andyross, 2021-02-03:
Cleaning up.  This is a duplicate of #31726 and the clock changed in #31841 (which can be merged, but is really a workaround that needs to be matched with simulator config) resolves it.

closed 2021-02-03 by andyross