zephyrproject-rtos/zephyr #31555: drivers_can_api: mimxrt1060 can api test meet assert failure

hakehuang, 2021-01-23:
**Describe the bug**
test_filter_attach ASSERTION FAIL [!FLEXCAN_IsMbOccupied(base, mbIdx)] @ WEST_TOPDIR/modules/hal/nxp/mcux/drivers/imx/fsl_flexcan.c:1734 E: r0/a1: 0x00000004 r1/a2: 0x000006c6 r2/a3: 0x00800000 E: r3/a4: 0x60004121 r12/ip: 0x00000000 r14/lr: 0x600064a3 E: xpsr: 0x41000000 E: Faulting instruction address (r15/pc): 0x6000a876 E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0 E: Current thread: 0x80000218 (unknown) E: Halting system </error>

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=mimxrt1060_evk ..
3. make flash --runner=pyocd
4. See error

**Expected behavior**
test sucessfully

**Impact**
can operation

**Logs and console output**
```
test_filter_attach ASSERTION FAIL [!FLEXCAN_IsMbOccupied(base, mbIdx)] @ WEST_TOPDIR/modules/hal/nxp/mcux/drivers/imx/fsl_flexcan.c:1734 E: r0/a1: 0x00000004 r1/a2: 0x000006c6 r2/a3: 0x00800000 E: r3/a4: 0x60004121 r12/ip: 0x00000000 r14/lr: 0x600064a3 E: xpsr: 0x41000000 E: Faulting instruction address (r15/pc): 0x6000a876 E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0 E: Current thread: 0x80000218 (unknown) E: Halting system </error>
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, )
 - Toolchain (Zephyr SDK, ...)
 - Commit SHA or Version used:
 zephyr-v2.4.0-3224-gc77ae15c9967

hakehuang, 2021-01-23:
@MaureenHelm  this looks like a test application settings issue, I will try to fix it in test side

hakehuang, 2021-01-23:
@henrikbrixandersen  do you test this driver with can_api before?

henrikbrixandersen, 2021-01-23:
@hakehuang If you are referring to the FlexCAN shim driver, then yes. I have just retested it along with the Kinetis FlexCAN driver on `frdm_k64f`:

```
*** Booting Zephyr OS build zephyr-v2.4.0-3549-g4ac2c420f9e6  ***
Running test suite can_driver
===================================================================
START - test_set_loopback
 PASS - test_set_loopback
===================================================================
START - test_send_and_forget
 PASS - test_send_and_forget
===================================================================
START - test_filter_attach
 PASS - test_filter_attach
===================================================================
START - test_receive_timeout
 PASS - test_receive_timeout
===================================================================
START - test_send_callback
 PASS - test_send_callback
===================================================================
START - test_send_receive_std
 PASS - test_send_receive_std
===================================================================
START - test_send_receive_ext
 PASS - test_send_receive_ext
===================================================================
START - test_send_receive_std_masked
 PASS - test_send_receive_std_masked
===================================================================
START - test_send_receive_ext_masked
 PASS - test_send_receive_ext_masked
===================================================================
START - test_send_receive_buffer
 PASS - test_send_receive_buffer
===================================================================
START - test_send_receive_wrong_id
 PASS - test_send_receive_wrong_id
===================================================================
START - test_send_invalid_dlc
E: DLC of 9 exceeds maximum (8)
 PASS - test_send_invalid_dlc
===================================================================
Test suite can_driver succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL
```

I have not tested the i.MX FlexCAN driver variant as I do not have access to any i.MX boards.

hakehuang, 2021-01-25:
@henrikbrixandersen , Thanks this issue only happens on RT1xxx series boards, let me check it further

hakehuang, 2021-01-25:
the problem is due to RT SOC ERRATA
```
#if ((defined(FSL_FEATURE_FLEXCAN_HAS_ERRATA_5641) && FSL_FEATURE_FLEXCAN_HAS_ERRATA_5641) || \
     (defined(FSL_FEATURE_FLEXCAN_HAS_ERRATA_5829) && FSL_FEATURE_FLEXCAN_HAS_ERRATA_5829))
        if (0U == mbIdx)
        {
            fgRet = true;
        }
        else
#endif
        {
            fgRet = false;
        }
    }
```
so when the test application use id 0 then it will return true which mean 0 is engaged.

closed 2021-02-01 by nashif