zephyrproject-rtos/zephyr #31566: up_squared: Couldn't get testcase log from console for all testcases.

chen-png, 2021-01-24:
**Description**
with the latest mainline branch, when running tests on up_squared board, couldn't get testcase log from serial for all testcases, there is nothing in console output.
the commit 6ec4ff22e3 is good to get log.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b up_squared -p auto samples/hello_world/
2. west flash ~/up_squared.sh
3. see error

**Logs and console output**

**nothing in console output**

**Environment (please complete the following information):**
 - OS: Fedora32
 - Toolchain: zephyr-sdk-0.12.0
 - Commit ID: 77861037d9c

chen-png, 2021-01-24:
git bisect and found the first bad commit is 5c47bbc5010d98c6

dcpleung, 2021-01-25:
Please re-test. This should have already been fixed by https://github.com/zephyrproject-rtos/zephyr/pull/31564

chen-png, 2021-01-25:
> Please re-test. This should have already been fixed by #31564

this test commit 7786103 already include #31564

chen-png, 2021-01-25:
@dcpleung how do you run tests on up_squared board, using usb disk, is it good to get log? for QA automation test, it used network to transfer image, nothing in the output.

dcpleung, 2021-01-25:
I compile the app on UP2, copy them into boot partition, then boot.

chen-png, 2021-01-26:
@dcpleung you could use the remote up_squared in SH lab, it could always reproduce this issue

tbursztyka, 2021-01-26:
I have the same issue, and a git bisect points to commit 5c47bbc 
My UP² is the celeron version (not the atom one). I had already found difference in the past like the CPU's apic ids (which had to be grabbed from ACPI to get things running)

@dcpleung so you use the efi way, not the multiboot (with grub)?

dcpleung, 2021-01-26:
I could only constantly boot with GRUB, and it still works with latest master.

EFI boots on my Atom UP2 if I disable `CONFIG_PCIE_MMIO_CFG` before the paging PR. But now this trick does not work anymore.

tbursztyka, 2021-01-26:
@dcpleung ok, again some difference between atom and celeron then. Because on my side both EFI and GRUB fail.

dcpleung, 2021-01-26:
My suggestion is to try with `CONFIG_PCIE_MMIO_CFG=n` and see how it goes.

dcpleung, 2021-01-26:
Could you try https://github.com/zephyrproject-rtos/zephyr/pull/31634 and see if that fixes the issue?

chen-png, 2021-01-27:
tested with the #31634 , when building testcase on up_squared, it will report a new build error as below:

cmd: west build -b up_squared -p auto samples/hello_world/
result: 
/home/ztest/zephyr-sdk-0.12.0/x86_64-zephyr-elf/bin/**x86_64-zephyr-elf-objcopy: zefi.elf: invalid bfd target**
Traceback (most recent call last):
  File "/home/ztest/work/zephyrproject/zephyr/arch/x86/zefi/zefi.py", line 155, in <module>
    build_elf(args.elf_file)
  File "/home/ztest/work/zephyrproject/zephyr/arch/x86/zefi/zefi.py", line 134, in build_elf
    subprocess.run(cmd, check = True)
  File "/usr/lib/python3.6/subprocess.py", line 438, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['/home/ztest/zephyr-sdk-0.12.0/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-objcopy', '--target=efi-app-x86_64', '-j', '.text', '-j', '.reloc', '-j', '.data', 'zefi.elf', 'zephyr.efi']' returned non-zero exit status 1.
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake --build /home/ztest/work/zephyrproject/zephyr/build

dcpleung, 2021-01-27:
This is caused by the SDK. You need SDK 0.12.**2**.

closed 2021-01-27 by nashif

chen-png, 2021-01-27:
thanks, it's good now with sdk 0.12.2.

chen-png, 2021-01-27:
tested and this issue was fixed by #31634. 
However, with the latest commit 51c34bb6, this error happened again, and git bisect, found the first bad commit is 51c34bb6.

chen-png, 2021-01-28:
> However, with the latest commit [51c34bb](https://github.com/zephyrproject-rtos/zephyr/commit/51c34bb60986d5db0286bf59eb10303ab10ef39c), this error happened again, and git bisect, found the first bad commit is [51c34bb](https://github.com/zephyrproject-rtos/zephyr/commit/51c34bb60986d5db0286bf59eb10303ab10ef39c).

this has been fixed in fda94e79caf9

PR #31634 (2021-01-26): x86: use large VM size if ACPI
We've already enabled full RAM mapping if ACPI is enabled, also
set a large 3GB address space size, these systems are not RAM-
constrained (they are PC platforms) and they have large MMIO
config spaces for PCIe.

Fixes: #31631
Fixes: #31566