zephyrproject-rtos/zephyr #31569: lora: sx126x: interrupt pin permanently enabled

JordanYates, 2021-01-24:
**Describe the bug**
The sx126x driver leaves the `DIO1` interrupt permanently enabled, resulting in excess power consumption.
For nRF SoC's, this means the `GPIOTE` peripheral is permanently enabled, increasing current draw by ~15uA.
Per the datasheet (Table 8-3), DIO1 is not driven as an output in `Reset`, `Startup`, or `Sleep` modes.
We can therefore safely turn off the interrupt in sleep mode.

**Expected behavior**
`DIO1` pin should only have the interrupt enabled when it is actually needed (Not in sleep mode).

**Impact**
Minor, permanent +15uA to the current consumption of any application.

**Environment (please complete the following information):**
 - Zephyr v2.4

tagunil, 2021-01-24:
Does the sx1276 driver have the same issue?

JordanYates, 2021-01-24:
> Does the sx1276 driver have the same issue?

Yes, all DIO pins have their interrupts permanently enabled in `SX1276IoIrqInit`.
The sx1276 has a much different sleep model, so the same solution doesn't directly apply.
That driver also currently uses `loramac-node` implementations for sleep and standby transitions.
I don't have any hardware with the older LoRa modem however, so I can't validate any potential solutions.

PR #31572 (2021-02-10): lora: power consumption fixes
Fixes two causes of excess power consumption.

The initial commit keeps the drivers power state variable in sync with the actual state, to ensure the wakeup function is only called once per exit from sleep.

The second commit disables the DIO1 interrupt when the modem is in sleep mode, fixes #31569

The third commit moves when the LoRaWAN MAC layer is notified of the public network state.
The new location in `lorawan_join` is the more correct location for the command, and reduces power consumption when `lorawan_start` and `lorawan_join` are separated in time.
commit e756ad6c (2021-02-10): lora: sx126x: disable DIO1 in sleep mode

Disable the DIO1 interrupt when the sx126x modem is in sleep mode.
On nRF hardware, this lets the `GPIOTE` hardware be switched off, saving
15uA.

Fixes #31569.
closed 2021-02-10 by nashif