zephyrproject-rtos/zephyr #31573: Wrong log settings in can_stm32 driver

Nukersson, 2021-01-24:
https://github.com/zephyrproject-rtos/zephyr/blob/77861037d9ca71f0093c3e604b7545084ce7fb16/drivers/can/can_stm32.c#L20

It must be `LOG_MODULE_REGISTER` instead of `LOG_MODULE_DEFINE`.

alexanderwachter, 2021-01-24:
LOG_MODULE_REGISTER(can_driver);
Is called in can_common.c:13.
You can only register a module once.

Nukersson, 2021-01-24:
> LOG_MODULE_REGISTER(can_driver);
> Is called in can_common.c:13.
> You can only register a module once.

LOG_MODULE_DEFINE not worked for me with samples/can/driver

alexanderwachter, 2021-01-24:
the sample builds without error on the latest HEAD.
`west build -b nucleo_f746zg zephyr/samples/drivers/can`

Nukersson, 2021-01-24:
> nucleo_f746zg

Sorry. It was my bad. It is not worked for me with my h7 can driver under dev.

Nukersson, 2021-01-24:
Found it out.

I've used `LOG_MODULE_DECLARE(can_stm32h7_driver, CONFIG_CAN_LOG_LEVEL); ` with different name than `can_driver`, thus I need `LOG_MODULE_DEFINE` since it is not declared in can_common.c

closed 2021-01-24 by Nukersson

alexanderwachter, 2021-01-24:
BTW @Nukersson #31061should work on the H7 too.

Nukersson, 2021-01-24:
> BTW @Nukersson #31061 should work on the H7 too.

Thank you very much! I will try to help in PR pointed by you.