zephyrproject-rtos/zephyr #31579: sam_e70_xplained: running tests/subsys/logging/log_core failed

Zhaoningx, 2021-01-25:
Steps to reproduce the behavior:
twister -p sam_e70_xplained --device-testing --device-serial /dev/ttyACM1 -T tests/subsys/logging/log_core
see error
*** Booting Zephyr OS build zephyr-v2.4.0-3601-g7191b64c6f08  ***
Running test suite test_log_list

===================================================================
START - test_log_backend_runtime_filtering
Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/logging/log_core/src/log_core_test.c:65: put: (log_msg_timestamp_get(msg) not equal to exp_timestamp)
Unexpected message index
FAIL - test_log_backend_runtime_filtering

===================================================================
START - test_log_overflow
Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/logging/log_core/src/log_core_test.c:65: put: (log_msg_timestamp_get(msg) not equal to exp_timestamp)
Unexpected message index
FAIL - test_log_overflow

===================================================================
START - test_log_arguments
Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/logging/log_core/src/log_core_test.c:301: test_log_arguments: (8 not equal to backend1_cb.counter)
Unexpected amount of messages received by the backend.
FAIL - test_log_arguments

===================================================================
START - test_log_from_declared_module
Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/logging/log_core/src/log_core_test.c:325: test_log_from_declared_module: (2 not equal to backend1_cb.counter)
Unexpected amount of messages received by the backend.
FAIL - test_log_from_declared_module

===================================================================
START - test_log_strdup_gc
Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/logging/log_core/src/log_core_test.c:352: test_log_strdup_gc: (2 not equal to backend1_cb.counter)
Unexpected amount of messages received by the backend.
FAIL - test_log_strdup_gc

===================================================================
START - test_log_strdup_detect_miss
PASS - test_log_strdup_detect_miss

===================================================================
START - test_strdup_trimming
Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/logging/log_core/src/log_core_test.c:434: test_strdup_trimming: (1 not equal to backend1_cb.counter)
Unexpected amount of messages received by the backend.
FAIL - test_strdup_trimming

===================================================================
START - test_log_msg_dropped_notification
Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/logging/log_core/src/log_core_test.c:449: log_n_messages: (backend1_cb.total_drops not equal to exp_dropped)
Unexpected log msg dropped 24 (expected 0)
FAIL - test_log_msg_dropped_notification

===================================================================
START - test_z_log_get_s_mask
PASS - test_z_log_get_s_mask

===================================================================
START - test_log_panic
Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/logging/log_core/src/log_core_test.c:517: test_log_panic: (2 not equal to backend1_cb.counter)
Unexpected amount of messages received by the backend.
FAIL - test_log_panic

===================================================================
Test suite test_log_list failed.

===================================================================
PROJECT EXECUTION FAILED

Environment (please complete the following information):

OS: Fedora33
Toolchain: zephyr-sdk-0.12.0
Commit ID: 7191b64c6f08

pabigot, 2021-01-25:
@Zhaoningx Do you have the sam_e70_xplained or the sam_e70b_xplained?  The former fails for me on my E70B, the latter passes.

Look at the big chip.  The E70B has:

         Atmel
      ATSAME70Q21
           AN
            B

nashif, 2021-01-26:
@Zhaoningx  ping

Zhaoningx, 2021-01-27:
Hi @nashif  @pabigot , I'm sorry, I have reported this before, and nashif has tested successfully, so I will find another e70 board to test this, so closed first. see https://github.com/zephyrproject-rtos/zephyr/issues/31340

Zhaoningx, 2021-02-02:
The bug is still existing, so re-open this issue.

pabigot, 2021-02-02:
Are you sure you don't have a sam_e70b_xplained?  That is not the same board as sam_e70_xplained, though that's what it's called.

Zhaoningx, 2021-02-02:
Hi @pabigot , yep, I have sam_e70_xplained board, it will failed when I run tests/subsys/logging/log_core, and I used another sam_e70_xplained board, the bug is still existing.

pabigot, 2021-02-02:
Run the same test with `sam_e70b_xplained`.

Zhaoningx, 2021-02-02:
You are right, the test is passed using sam_e70b_xplained

nashif, 2021-02-02:
@Zhaoningx please confirm what board you have exactly, this is starting to get confusing. I have rev A and the test passes just fine.

nandojve, 2021-02-02:
Just to clarify, A and B means IC revision and not board revision. So, B IC's fixes some A problems and need look at errata docs.
In fact, you can buy a B LQFP and replace A chip in the DevKit.

[image]

to make sure:
[image]

Zhaoningx, 2021-02-03:
Hi @nashif @nandojve , thanks for your comments, it's my mistake and my board is sam_e70b_xplained not sam_e70_xplained, and it passed to test, so closed this.

closed 2021-02-03 by Zhaoningx