zephyrproject-rtos/zephyr #31582: STM32F746ZG: No pwm signal output when running /tests/drivers/pwm/pwm_api

gitificial, 2021-01-25:
**Describe the bug**
I'm not sure if it's a bug or if I'm missing something in my settings.
When running the pwm_api test on my nucleo_f746zg the minicom output shows a successful test run (see output below). But when I measure the output of pin PE13 (as defined in the generated zephyr.dts file) the output voltage resides on a constant level. Simple toggling of this pin with the GPIO methods work flawlessly.
Once I changed CONFIG_PWM=y to CONFIG_PWM_STM32=y in the prj.conf file but I got errors with this setting. 

**To Reproduce**
Steps to reproduce the behavior:
1. cd ~/zephyrproject/zephyr
2. west build -p auto -b nucleo_f746zg /home/user/zephyrproject/zephyr/tests/drivers/pwm/pwm_api/
3. west flash
4. sudo minicom -D /dev/ttyACM0 -o
5. while the test is running: measuring output voltage on pin PE13 (D3) with a DSO -> trigger "normal" @2V2

**Expected behavior**
PWM signal on PE13 while the test is running.

**Impact**
showstopper

**Logs and console output**
Minicom output:

Running test suite pwm_basic_test
===================================================================
START - test_pwm_usec
[PWM]: 1, [period]: 2000, [pulse]: 1000
[PWM]: 1, [period]: 2000, [pulse]: 2000
[PWM]: 1, [period]: 2000, [pulse]: 0
 PASS - test_pwm_usec
===================================================================
START - test_pwm_nsec
[PWM]: 1, [period]: 2000000, [pulse]: 1000000
[PWM]: 1, [period]: 2000000, [pulse]: 2000000
[PWM]: 1, [period]: 2000000, [pulse]: 0
 PASS - test_pwm_nsec
===================================================================
START - test_pwm_cycle                                                          
[PWM]: 1, [period]: 64000, [pulse]: 32000                                       
[PWM]: 1, [period]: 64000, [pulse]: 64000                                       
[PWM]: 1, [period]: 64000, [pulse]: 0                                           
 PASS - test_pwm_cycle                                                          
===================================================================             
Test suite pwm_basic_test succeeded                                             
===================================================================             
PROJECT EXECUTION SUCCESSFUL 

**Environment (please complete the following information):**
 - Ubuntu 20.04.1 LTS
 - Zephyr 2.4.99
 - zephyr-sdk-0.11.4

**Additional context**
- used DSO with 1GS/s sampling rate

erwango, 2021-01-25:
@gmarull would you have time to have a look ?

erwango, 2021-01-26:
@gitificial Test requires a small tweak to use pwm channel assigned to PE13:
```
 #elif DT_HAS_COMPAT_STATUS_OKAY(st_stm32_pwm)
-#define DEFAULT_PWM_PORT 1
+#define DEFAULT_PWM_PORT 3
 #else
```

gmarull, 2021-01-26:
PWM test has a hard-coded channel (1) for STM32, so if using something else it needs to be changed as @erwango suggests.

closed 2021-01-26 by ioannisg

gitificial, 2021-01-26:
@erwango: Thank you! #define DEFAULT_PWM_PORT 3 solved it for me. I did a fallacy regarding the pin assignments.

PR #31612 (2021-01-26): tests/drivers: pwm_api: Document how to tune test for stm32 targets
The way the test is implemented requires some adaptation to
be effective on all STM32 boards.
I'm not adding per board code on purpose, as this would be a
never ending story.

Fixes #31582
commit 3d267eee (2021-01-26): tests/drivers: pwm_api: Document how to tune test for stm32 targets

The way the test is implemented requires some adaptation to
be effective on all STM32 boards.
I'm not adding per board code on purpose, as this would be a
never ending story.

Fixes #31582