zephyrproject-rtos/zephyr #31599: 64 bit race on timer counter in cavs_timer

andyross, 2021-01-25:
The CAVS hardware has a 64 bit timer counter (it's the same device as the HD Audio wall clock timer).  But the code to read it looks like this:

```c
static uint64_t count(void)
{
	return shim_regs->walclk;
}
```

When compiled on Xtensa, this produces two sequential 32 bit reads of the value.  And that's wrong, because the hardware continues to run.  It's possible for the two reads to straddle a rollover of the low word and increment of the high word.

This needs to be fixed to be safe.  The simplest scheme is to read the high word twice, compare the two values, and repeat the process if they don't match.

andyross, 2021-01-25:
Discovered via conversation with @lgirdwood  and @lyakh who had tripped over the same bug in upstream (non-Zephyr) SOF code.  We have exactly the same messup.

lgirdwood, 2021-01-26:
@andyross fell free to copy and paste from SOF once it merged and passes QA and validation.

PR #31958 (2021-02-04): drivers/cavs_timer: Fix multiword race with timer counter
The count register is 64 bits, but we're a 32 bit CPU that can only
read four bytes at a time, so a bit of care is needed to prevent
racing against a wraparound of the low word.  Wrap the low read
between two reads of the high word and make sure it didn't change.

Fixes #31599
commit 71fd58cc (2021-02-04): drivers/cavs_timer: Fix multiword race with timer counter

The count register is 64 bits, but we're a 32 bit CPU that can only
read four bytes at a time, so a bit of care is needed to prevent
racing against a wraparound of the low word.  Wrap the low read
between two reads of the high word and make sure it didn't change.

Fixes #31599
closed 2021-02-04 by nashif