zephyrproject-rtos/zephyr #31613: Undefined reference errors when using External Library with k_msgq_* calls

MaikVermeulen, 2021-01-26:
**Describe the bug**
I made a very simple library that includes two functions that use the message queue. However, I get this linking error:
`undefined reference to `z_impl_k_msgq_put'`

I think it's due to ``z_impl_k_msgq_put'` being declared as extern and that affects the linking, but I haven't found a solution. I tried replacing the `k_msgq_*` functions with `LOG_INF`, and then it works just fine.

I'm trying to put some code into libraries so that we can share those with external parties, without sharing our source code with them.

**To Reproduce**
[I have created a simple minimal example of the problem.](https://github.com/MaikVermeulen/zephyr_external_library_issue)

**Expected behavior**
A clean build.

**Impact**
I can't librarize Zephyr code that includes calls to the message queue (and possibly similar code), hence I can't provide external parties with the modules they need without sharing our source code.

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: Windows
 - nRF Connect SDK v1.4.99-dev1

pabigot, 2021-01-26:
Please look at 6b720cb3c56a198db19b1884e78385cb7a392b00 for a possible fit.

MaikVermeulen, 2021-01-27:
@pabigot This pointed me in the right direction, thank you.
Simply linking the kernel library resulted in a strong component cycle error, so I included the dummy library as PRIVATE:
`zephyr_library_link_libraries(kernel)
target_link_libraries(app PRIVATE lib)`

Now it works. Thanks!

closed 2021-01-27 by MaikVermeulen

MaikVermeulen, 2021-05-12:
@danderson-presonus My `CMakeLists.txt` in the library folder containing the `.a` file:
```
zephyr_library_import(lib ${CMAKE_CURRENT_SOURCE_DIR}/liblib.a)
target_include_directories(app PUBLIC ${CMAKE_CURRENT_SOURCE_DIR})
target_link_libraries(lib INTERFACE kernel)
target_link_libraries(app PRIVATE lib)
```

danderson-presonus, 2021-05-12:
Thanks!  This got me up and running.

CManigandan, 2022-05-18:
> Please look at 6b720cb3c56a198db19b1884e78385cb7a392b00 for a possible fix.

@pabigot I am not able to locate this fix. Can you help?