zephyrproject-rtos/zephyr #31616: test: ipc: Test from samples/subsys/ipc/rpmsg_service fails on nrf5340dk_nrf5340_cpuapp

PerMac, 2021-01-26:
**Describe the bug**
The test from the title fails on nrf53_cpuapp. The sample is started and hangs after printing:
DEVICE: RPMsg Service [master] demo started
 
**To Reproduce**
Steps to reproduce the behavior:
1. Have nrf5340dk connected
2. call `/scripts/twister --device-testing -T samples/subsys/ipc/rpmsg_service/ -p nrf5340dk_nrf5340_cpuapp --device-serial /dev/ttyACM2 -v -v`
3. See error

**Expected behavior**
The test passes

**Impact**
Not clear

**Logs and console output**
```
DEBUG   - DEVICE: *** Booting Zephyr OS build v2.5.0-rc1-16-g085d2eda8d2b  ***
DEBUG   - DEVICE: Starting application thread!
DEBUG   - DEVICE: 
DEBUG   - DEVICE: RPMsg Service [master] demo started
```
and then hangs

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain Zephyr SDK
 - Commit SHA or Version used: v2.5.0-rc1-16-g085d2eda8d2b

ioannisg, 2021-02-03:
will take a look

ioannisg, 2021-02-04:
@PerMac I am not sure this is a bug. For the sample to work one needs to flash the two cores (master and remote), I guess, separately. West build takes care of building both executables, but the images needs to be flashed individually.

closed 2021-02-04 by nashif

KeremSAR, 2023-04-06:
I have the same issue, It hangs there for both remote and master application.
`while (!rpmsg_service_endpoint_is_bound(ep_id)) {
		k_sleep(K_MSEC(1));
	}`
I flashed the images individually for nrf5340 cores. What could be the issue ?

pauleffect90, 2024-02-10:
Same issue. @KeremSAR. Did you manage to fix it?

Thomas000001, 2025-03-03:
Hello, @KeremSAR and @pauleffect90, I also got the same issue of  while (!rpmsg_service_endpoint_is_bound(ep_id)) { k_sleep(K_MSEC(1)); } . Did you fix it?
Sincerely.

pauleffect90, 2025-03-03:
> Hello, [@ioannisg](https://github.com/ioannisg) 、[@KeremSAR](https://github.com/KeremSAR) and [@pauleffect90](https://github.com/pauleffect90), I also got the same issue of while (!rpmsg_service_endpoint_is_bound(ep_id)) { k_sleep(K_MSEC(1)); } . Did you fix it? Sincerely.

yeah, from what I remember, it was an overlay issue. The tx/ rx endpoint addresses (ram) need to be inverted on the app/ network core (like the uart rx goes to the uart tx). Im speaking off the top of my head, though.

Thomas000001, 2025-03-04:
@pauleffect90 OK, I will try it later. Thank you for your suggestion.

PR #31997 (2021-02-04): samples: rpmsg_service: do not run test for nrf5340 in CI
We cannot automatically flash both master and remote images
to nRF5340 application and network core, this needs to be
done separately. We therefore, cannot execute the test on
nRF5340, so we switch to make the test build-only on this
platform.

Fixes #31616