zephyrproject-rtos/zephyr #31627: tests/subsys/power/power_mgmt/subsys.power.device_pm  fails to build on nrf5340dk_nrf5340_cpunet &  nrf5340pdk_nrf5340_cpunet

galak, 2021-01-26:
```
scripts/twister -p nrf5340dk_nrf5340_cpunet -p nrf5340pdk_nrf5340_cpunet -s tests/subsys/power/power_mgmt/subsys.power.device_pm
```

galak, 2021-01-26:
@carlescufi please re-assign this as needed.

nashif, 2021-01-27:
@ceolin 
you should not be filter hardware in subsystems, this needs to be done on the test level 

```diff
diff --git a/subsys/power/Kconfig b/subsys/power/Kconfig
index 84e180ffca..c3e676da6f 100644
--- a/subsys/power/Kconfig
+++ b/subsys/power/Kconfig
@@ -13,7 +13,7 @@ config SYS_POWER_MANAGEMENT
 menuconfig PM
        bool "System Power management"
        select TICKLESS_IDLE
-       depends on SYS_CLOCK_EXISTS && !SOC_NRF5340_CPUNET_QKAA
+       depends on SYS_CLOCK_EXISTS
        help
          This option enables the board to implement extra power management
          policies whenever the kernel becomes idle. The kernel informs the
```

nashif, 2021-01-27:
this boils down to this board failing when CONFIG_PM is enabled (which enables build of `soc/arm/nordic_nrf/nrf53/power.c` with the following failure:

```
/home/nashif/Work/zephyrproject/modules/hal/nordic/nrfx/hal/nrf_regulators.h:138:50: error: unknown type name 'NRF_REGULATORS_Type'; did you mean 'NRF_REGULATORS_H__'?
  138 | NRF_STATIC_INLINE void nrf_regulators_dcdcen_set(NRF_REGULATORS_Type * p_reg, bool enable);
      |                                                  ^~~~~~~~~~~~~~~~~~~
      |                                                  NRF_REGULATORS_H__
/home/nashif/Work/zephyrproject/modules/hal/nordic/nrfx/hal/nrf_regulators.h:150:50: error: unknown type name 'NRF_REGULATORS_Type'; did you mean 'NRF_REGULATORS_H__'?
  150 | NRF_STATIC_INLINE void nrf_regulators_system_off(NRF_REGULATORS_Type * p_reg);
      |                                                  ^~~~~~~~~~~~~~~~~~~
      |                                                  NRF_REGULATORS_H__
/home/nashif/Work/zephyrproject/modules/hal/nordic/nrfx/hal/nrf_regulators.h:234:50: error: unknown type name 'NRF_REGULATORS_Type'; did you mean 'NRF_REGULATORS_H__'?
  234 | NRF_STATIC_INLINE void nrf_regulators_dcdcen_set(NRF_REGULATORS_Type * p_reg, bool enable)
      |                                                  ^~~~~~~~~~~~~~~~~~~
      |                                                  NRF_REGULATORS_H__
/home/nashif/Work/zephyrproject/modules/hal/nordic/nrfx/hal/nrf_regulators.h:243:50: error: unknown type name 'NRF_REGULATORS_Type'; did you mean 'NRF_REGULATORS_H__'?
  243 | NRF_STATIC_INLINE void nrf_regulators_system_off(NRF_REGULATORS_Type * p_reg)
      |                                                  ^~~~~~~~~~~~~~~~~~~
      |                                                  NRF_REGULATORS_H__
/home/nashif/Work/zephyrproject/zephyr/soc/arm/nordic_nrf/nrf53/power.c: In function 'pm_power_state_set':
/home/nashif/Work/zephyrproject/zephyr/soc/arm/nordic_nrf/nrf53/power.c:20:3: warning: implicit declaration of function 'nrf_regulators_system_off' [-Wimplicit-function-declaration]
   20 |   nrf_regulators_system_off(NRF_REGULATORS);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/nashif/Work/zephyrproject/zephyr/soc/arm/nordic_nrf/nrf53/power.c:20:29: error: 'NRF_REGULATORS' undeclared (first use in this function); did you mean 'NRF_REGULATORS_H__'?
   20 |   nrf_regulators_system_off(NRF_REGULATORS);
      |                             ^~~~~~~~~~~~~~
      |                             NRF_REGULATORS_H__
/home/nashif/Work/zephyrproject/zephyr/soc/arm/nordic_nrf/nrf53/power.c:20:29: note: each undeclared identifier is reported only once for each function it appears in
```

Should building of power.c in the SoC depend on something else to prevent that or to correctly pull in regulator support? @pabigot  ?

pabigot, 2021-01-27:
patch incoming.

ceolin, 2021-01-27:
> @ceolin
> you should not be filter hardware in subsystems, this needs to be done on the test level
>
> ```diff
> diff --git a/subsys/power/Kconfig b/subsys/power/Kconfig
> index 84e180ffca..c3e676da6f 100644 [...]

gotcha, but IMHO tit is not just exclude this target from the test as the proposed patch, but also remove CONFIG_PM for this target.

ceolin, 2021-01-27:
https://github.com/zephyrproject-rtos/zephyr/issues/31627 should complement #31714

PR #31714 (2021-02-01): fix build errors with nr53 net core and power management
The net core of nRF53 doesn't provide access to the power regulator features: they can be accessed only from the application core.

Revert the previous way of attempting to avoid the test problem, and exclude the net cores from the test.

Fixes #31627
closed 2021-02-01 by nashif