zephyrproject-rtos/zephyr #31640: mcuboot build is broken

gmarull, 2021-01-26:
**Describe the bug**

mcuboot build is broken with current Zephyr master.

**To Reproduce**

```
west build -b nrf52840dk_nrf52840 -d build_bootloader -s ../bootloader/mcuboot/boot/zephyr
```

**Expected behavior**

Build succeeds

**Impact**

mcuboot bootloader can't be built.

**Logs and console output**

Relevant CMake errors:

```
error: LOG_MINIMAL (defined at subsys/logging/Kconfig.mode:51) is assigned in a configuration file,
but is not directly user-configurable (has no prompt). It gets its value indirectly from other
symbols. See http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_LOG_MINIMAL.html and/or
look up LOG_MINIMAL in the menuconfig/guiconfig interface. The Application Development Primer,
Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be
helpful too.
```

**Environment (please complete the following information):**

 - OS: Ubuntu 20.04
 - Toolchain: Zephyr SDK 0.11.4
 - Commit SHA or Version used: 4720585077399f09cbb1314627ecac3d2ad45be4

**Additional context**

Reverting https://github.com/zephyrproject-rtos/zephyr/pull/31404 fixes the problem

henrikbrixandersen, 2021-01-26:
Reverting #31404 should not be necessary. The MCUboot Zephyr just needs to be updated to use the new Kconfig options.

gmarull, 2021-01-26:
@henrikbrixandersen I agree, but until then Zephyr master should be in a working state

gmarull, 2021-01-26:
Opened https://github.com/mcu-tools/mcuboot/pull/932

nandojve, 2021-01-27:
Question: How is possible MCUboot be broken at master? Are tests running to avoid this?
This is not an exception, I believe, and I would like to see something addressed to avoid this in Future.

nvlsianpu, 2021-01-27:
> Question: How is possible MCUboot be broken at master? Are tests running to avoid this?
This is not an exception, I believe, and I would like to see something addressed to avoid this in Future.

@nandojve https://github.com/zephyrproject-rtos/zephyr/issues/29397
saw that problem, not fixed so fare

PR #31710 (2021-01-27): manifest: fix MCUboot build after #31404
MCUboot version with bugfix for following issue:
CONFIG_LOG_MINIMAL was replaced by
CONFIG_LOG_MODE_MINIMAL by
zephyrproject-rtos/zephyr#31404 which
broke MCUboot default build.

fixes #31640
commit 830eb582 (2021-01-27): manifest: fix MCUboot build after #31404

MCUboot version with bugfix for following issue:
CONFIG_LOG_MINIMAL was replaced by
CONFIG_LOG_MODE_MINIMAL by
zephyrproject-rtos/zephyr#31404 which
broke MCUboot default build.

fixes #31640
closed 2021-01-27 by nashif