zephyrproject-rtos/zephyr #31667: [Coverity CID :215395] Out-of-bounds access in tests/net/lib/dns_sd/src/main.c

zephyrbot, 2021-01-26:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/f91e9fba51e5da46ee5c6822f8656713d74a6ecf/tests/net/lib/dns_sd/src/main.c

Category: Memory - corruptions
Function: `test_add_ptr_record`
Component: Tests
CID: [215395](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=215395)

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

nashif, 2021-01-26:
false positive

cfriedt, 2021-02-01:
Marked as a false positive.
Already covered by 2 assertions directly above. assert actual_int > 0 and assert actual_int == expected_int.

closed 2021-02-01 by cfriedt