zephyrproject-rtos/zephyr #31674: [Coverity CID :216788] Explicit null dereferenced in tests/ztest/error_hook/src/main.c

zephyrbot, 2021-01-26:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/f91e9fba51e5da46ee5c6822f8656713d74a6ecf/tests/ztest/error_hook/src/main.c#L57

Category: Null pointer dereferences
Function: `trigger_fault_access_null`
Component: Tests
CID: [216788](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=216788)

Details:

```
51    
52     static void trigger_fault_access_null(void)
53     {
54         void *a = NULL;
55    
56         /* access a null of address */
>>>     CID 216788:  Null pointer dereferences  (FORWARD_NULL)
>>>     Dereferencing null pointer "a".
57         int b = *((int *)a);
58    
59         printk("b is %d\n", b);
60     }
61    
62     static void trigger_fault_divide_zero(void)
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

enjiamai, 2021-01-29:
@nashif  As you mentioned, this is intentionally dereferencing a null pointer to cause a crash for testing. Could you please close it in coverity, thanks.

pabigot, 2021-02-02:
2021-02-02 20:24:22 (change by @pabigot)
Classification:	Unclassified	→	Intentional
Comment:
This is supposed to dereference a null pointer.

closed 2021-02-02 by pabigot