zephyrproject-rtos/zephyr #31684: intel_adsp_cavs15: Cannot download firmware of kernel testcases

KangJianX, 2021-01-27:
**Describe the bug**
Use west to flash zephyr image on ADSP failed

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b intel_adsp_cavs15 tests/kernel/thread
2. west sign -t rimage -p ../modules/audio/sof/zephyr/ext/rimage/build/rimage -D ../modules/audio/sof/zephyr/ext/rimage/config/ -- -k ../modules/audio/sof/keys/otc_private_key.pem
3. /home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/fw_loader.py -f /home/ztest/work/zephyrproject/zephyr/build/zephyr/zephyr.ri

**Expected behavior**
Firmware load successful

**Logs and console output**
Start firmware downloading...
Open HDA device: /dev/hda
Reset DSP...
Firmware Status Register (0xFFFFFFFF)
   Boot:   0xFFFFFF (UNKNOWN)
   Wait:   0x0F (UNKNOWN)
   Module: 0x07
   Error:  0x01
   IPC CMD  : 0xFFFFFFFF
   IPC LEN  : 0xFFFFFFFF
Booting up DSP...
Firmware Status Register (0x05000001)
   Boot:   0x000001 (INIT_DONE)
   Wait:   0x05 (DMA_BUFFER_FULL)
   Module: 0x00
   Error:  0x00
Downloading firmware...
Traceback (most recent call last):
  File "/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/fw_loader.py", line 81, in <module>
    main()
  File "/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/fw_loader.py", line 63, in main
    fw_loader.download_firmware(args.firmware)
  File "/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/lib/loader.py", line 187, in download_firmware
    sd = self.load_firmware(fw_file)
  File "/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/lib/loader.py", line 174, in load_firmware
    sd.buf.copy(data, len(data))
  File "/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/lib/stream_desc.py", line 41, in copy
    self.buf[:] = data[:]
**ValueError: Can only assign sequence of same size**

**Environment** (please complete the following information):
OS: Fedora28
Toolchain: Zephyr-sdk-0.12.1
Commit ID: fc03bd2b863d543bd9

KangJianX, 2021-01-27:
Use twister has same problem, and show a fatal error not in the west log.

Log:
[5/6] cd /home/ztest/work/zephyrproject/zephyr/build/zephyr/cmake/flash && /usr/local/bin/cmake -E echo

-- west flash: using runner misc-flasher
Signing using default key
Signing for board intel_adsp_cavs15
Signing for SOC target apl using apl.toml
Signing with tool /usr/local/bin/rimage
**FATAL ERROR: Configuration not found**
/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/fw_loader.py -f /home/ztest/work/zephyrproject/zephyr/build/zephyr/zephyr.ri
Start firmware downloading...
Open HDA device: /dev/hda
Reset DSP...
Firmware Status Register (0xFFFFFFFF)
   Boot:   0xFFFFFF (UNKNOWN)
   Wait:   0x0F (UNKNOWN)
   Module: 0x07
   Error:  0x01
   IPC CMD  : 0xFFFFFFFF
   IPC LEN  : 0xFFFFFFFF
Booting up DSP...
Firmware Status Register (0x05000001)
   Boot:   0x000001 (INIT_DONE)
   Wait:   0x05 (DMA_BUFFER_FULL)
   Module: 0x00
   Error:  0x00
Downloading firmware...
Traceback (most recent call last):
  File "/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/fw_loader.py", line 81, in <module>
    main()
  File "/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/fw_loader.py", line 63, in main
    fw_loader.download_firmware(args.firmware)
  File "/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/lib/loader.py", line 187, in download_firmware
    sd = self.load_firmware(fw_file)
  File "/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/lib/loader.py", line 174, in load_firmware
    sd.buf.copy(data, len(data))
  File "/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/lib/stream_desc.py", line 41, in copy
    self.buf[:] = data[:]
**ValueError: Can only assign sequence of same size**

nashif, 2021-01-27:
@KangJianX please do not set priority.

closed 2021-02-01 by nashif