zephyrproject-rtos/zephyr #31696: UP² Celeron version (not the Atom one) has no console

tbursztyka, 2021-01-27:
**Describe the bug**
UP² Celeron either does not boot or at least lost its console output
(the early console already, since the boot banner is not even printed out)

**To Reproduce**
Use samples/hello_world  either on EFI or Grub

**Expected behavior**
Zephyr boot banner and hello world being printed out.

**Impact**
I use this board extensively and that is blocking my work atm 

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain:Zephyr SDK 12.2
 - Commit 51c34bb60986d5db0286bf59eb10303ab10ef39c

**Additional context**
I ran a git bisect and it is still the commit 5c47bbc5010d98c6f6d795aa6d268138c3463bbd

I have no idea why this commit affects Apollo Lake Celeron version and not the Atom one.

jhedberg, 2021-01-27:
@tbursztyka this might because of a rebasing mistake in #31618

Try removing the explicit KERNEL_VM_SIZE from defconfig and see if it helps

tbursztyka, 2021-01-27:
@jhedberg yep that's the issue, thanks for noticing this

tbursztyka, 2021-01-27:
So it was not only a celeron version issue (I though that master was working for up2 and as it did not with mine...).
Closing, #31703  is fixing it.

closed 2021-01-27 by tbursztyka